microbit-foundation/micropython-microbit-v2 #2: Update CODAL for public open source locations

jaustin, 2020-10-13:
Now that the new micro:bit is announced and the code is public, we have the repositories hosted at their correct, long-term home. We need to update the MicroPython build to reflect this

 - The repo that was 'codal' has moved to https://github.com/lancaster-university/microbit-v2-samples
 - The repo that was 'codal-microbit' has moved to https://github.com/lancaster-university/codal-microbit-v2

Other repositories have not moved

dpgeorge, 2020-10-14:
I tried updating it, but when I run the build script (from microbit-v2-samples) it tries to clone https://github.com/microbit-foundation/codal-microbit and asks for a password.  Maybe the target needs to be set?

microbit-carlos, 2020-10-14:
Gave this a go from a clean clone and couldn't replicate:

```
Creating libraries folder
Cloning into: https://github.com/lancaster-university/codal-microbit-v2
Cloning into 'codal-microbit-v2'...
Checking out branch: master
Already on 'master'
Your branch is up to date with 'origin/master'.
Set target: codal-microbit-v2 
Using target.json (dev version) 
Targeting codal-microbit-v2
-- The C compiler identification is GNU 7.3.1
-- The CXX compiler identification is GNU 7.3.1
-- Checking whether C compiler has -isysroot
-- Checking whether C compiler has -isysroot - yes
-- Checking whether C compiler supports OSX deployment target flag
-- Checking whether C compiler supports OSX deployment target flag - no
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking whether CXX compiler has -isysroot
-- Checking whether CXX compiler has -isysroot - yes
-- Checking whether CXX compiler supports OSX deployment target flag
-- Checking whether CXX compiler supports OSX deployment target flag - no
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /Users/microbit-carlos/opt/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-gcc
Supressing -Wexpansion-to-defined.
-- Adding library path: (/Users/microbit-carlos/workspace/deleteme/really-delete/microbit-v2-samples/libraries)
Installing dependencies...
Cloning into: https://github.com/lancaster-university/codal-core
Cloning into 'codal-core'...
Checking out branch: master
Already on 'master'
Your branch is up to date with 'origin/master'.
Cloning into: https://github.com/lancaster-university/codal-nrf52
Cloning into 'codal-nrf52'...
Submodule 'nrfx' (https://github.com/NordicSemiconductor/nrfx) registered for path 'nrfx'
Cloning into '/Users/microbit-carlos/workspace/deleteme/really-delete/microbit-v2-samples/libraries/codal-nrf52/nrfx'...
Submodule path 'nrfx': checked out '9136253e271f978a8da735864fc50d60e0ed68c6'
Checking out branch: master
Already on 'master'
Your branch is up to date with 'origin/master'.
Synchronizing submodule url for 'nrfx'
Cloning into: https://github.com/microbit-foundation/codal-microbit-nrf5sdk
Cloning into 'codal-microbit-nrf5sdk'...
Checking out branch: master
Already on 'master'
Your branch is up to date with 'origin/master'.
Using library: codal-microbit-v2
Using library: codal-core
Using library: codal-nrf52
Using library: codal-microbit-nrf5sdk
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/microbit-carlos/workspace/deleteme/really-delete/microbit-v2-samples/build
Scanning dependencies of target codal-core
```

Only foundation org repo is https://github.com/microbit-foundation/codal-microbit-nrf5sdk

Were you running that in a clean clone, or maybe in a directory that already had dependencies previously cloned  in the "libraries" folder?

dpgeorge, 2020-10-14:
Ok, thanks for that, I found my mistake, not updating the target in the custom codal.json file.

dpgeorge, 2020-10-14:
See #3 for an update.  I made it use the "codal" repository instead of "microbit-v2-samples" because it seemed cleaner.

dpgeorge, 2020-10-14:
Done in aec1c71794cbf97fbb88a101dc939654f30995e4

closed 2020-10-14 by dpgeorge