microbit-foundation/micropython-microbit-v2 #4: Neopixel Module Not Availabke

JoshuaLowe1002, 2020-10-15:
Apologies if you're already aware of this bug but it seems the neopixel module does not exist in MicroPython on V2.

Maybe an issue linked to https://github.com/microsoft/pxt-microbit/issues/3415

[image]

microbit-mark, 2020-11-03:
This module is also requested via micro:bit support in https://support.microbit.org/a/tickets/40468

martinwork, 2020-11-09:
@microbit-mark @microbit-carlos Needed for testing "display_and_neopixel.py"

microbit-carlos, 2020-11-09:
Yes, this is something we wanted to implement in the CODAL layer, so I've opened https://github.com/lancaster-university/codal-microbit-v2/issues/19.

jaustin, 2020-11-13:
The older implementation doesn't co-exist with Bluetooth so I think it does make sense to use the PWM-based one. @finneyj any thoughts on merging that so we can ensure Neopixel in MicroPython?  Would it require a dedicated PWM channel?

finneyj, 2020-11-13:
Yes, I agree we should build a PWM-based driver codal side. It does require a dedicated PWM peripheral, but I have one reserved for this already, so no problem.

Sam's driver needs a bit of work as I recall to package so that it is memory safe during DMA and double buffers to avoid memory overhead/glitching. I'll happily do that when I get a minute.

dpgeorge, 2020-11-16:
Ideally CODAL would expose a single function like `neopixel_write(pin, len, buf)` that would output the given `buf` of data on the `pin` using neopixel encoding.

microbit-carlos, 2020-11-16:
NeoPixel support has been added to CODAL: https://github.com/lancaster-university/codal-nrf52/blob/master/inc/neopixel.h
Thanks @finneyj !

finneyj, 2020-11-16:
no worries @dpgeorge @microbit-carlos.

Please note this is a blocking call (it will block the calling fiber until playout is complete), but doesn't disable interrupts or anything like that, so shouldn't have any of the side effects associated with the old bit banging implementations. 

Under the hood it all runs async, so see here if you prefer it to just run in the background. You just need to flip play() to playAsync(): 

https://github.com/lancaster-university/codal-nrf52/blob/master/source/neopixel.cpp#L25

https://github.com/lancaster-university/codal-nrf52/blob/master/source/WS2812B.cpp#L129

dpgeorge, 2020-11-17:
Thanks @finneyj !  A blocking call is good enough for now.

Implemented on the Python side in 915b6aa746ce35d40862b973c4dd519ff11109f2, should be compatible with micro:bit v1, and also includes support for RGBW LEDs.

closed 2020-11-17 by dpgeorge

microbit-carlos, 2020-11-17:
@microbit-mark @martinwork this should be ready for testing on the next MicroPython release.

microbit-carlos, 2020-11-17:
@dpgeorge the `ws2812_write` function is also exposed via `microbit` module (`microbit.ws2812_write`), is that expected?

dpgeorge, 2020-11-18:
> the `ws2812_write` function is also exposed via `microbit` module (`microbit.ws2812_write`), is that expected?

Yes that's expected, although at this stage I wouldn't document the function as it may change and/or be removed in the future.

We could move it out of the `microbit` module into something like `_microbit` intended for hidden functions if that is a better approach.