microbit-foundation/micropython-microbit-v2 #5: building firmware on debian10

leomlr, 2020-10-23:
Hi, I'm trying to build the firmware of the micro: bit v2 on Ubuntu 20.04 but it doesn't work for me. I need your 
help to solve this problem.

cmake version : 3.16.3
gcc-arm-none-eabi: 15:9-2019-q4-0ubuntu1

 I am getting this error during 'make' command in \src folder:
```
make -C codal_port
make[1] : on entre dans le répertoire « /home/leomlr/Documents/micropython-microbit-v2/src/codal_port »
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
python3 ../../lib/micropython/py/makeversionhdr.py build/genhdr/microbitversion.h
CC ../../lib/micropython/py/builtinimport.c
In file included from **../../lib/micropython/py/qstr.h:30**,
                 from **../../lib/micropython/py/lexer.h:32**,
                 from **../../lib/micropython/py/compile.h:29**,
                 from **../../lib/micropython/py/builtinimport.c:32**:
**../../lib/micropython/py/builtinimport.c**: In function '**mp_builtin___import__**':
./mpconfigport.h:35:50: error: 'PATH_MAX' undeclared (first use in this function)
   35 | #define MICROPY_ALLOC_PATH_MAX                  (PATH_MAX)
      |                                                  ^~~~~~~~
**../../lib/micropython/py/misc.h:169:63**: note: in definition of macro 'VSTR_FIXED'
  169 | vstr, alloc) vstr_t vstr; char vstr##_buf[(alloc)]; vstr_init_fixed_buf(&vstr, (alloc), vstr##_buf);
      |                                            ^~~~~

../../lib/micropython/py/builtinimport.c:360:22: note: in expansion of macro 'MICROPY_ALLOC_PATH_MAX'
  360 |     VSTR_FIXED(path, MICROPY_ALLOC_PATH_MAX)
      |                      ^~~~~~~~~~~~~~~~~~~~~~
./mpconfigport.h:35:50: note: each undeclared identifier is reported only once for each function it appears in
   35 | #define MICROPY_ALLOC_PATH_MAX                  (PATH_MAX)
      |                                                  ^~~~~~~~
../../lib/micropython/py/misc.h:169:63: note: in definition of macro 'VSTR_FIXED'
  169 | vstr, alloc) vstr_t vstr; char vstr##_buf[(alloc)]; vstr_init_fixed_buf(&vstr, (alloc), vstr##_buf);
      |                                            ^~~~~

../../lib/micropython/py/builtinimport.c:360:22: note: in expansion of macro 'MICROPY_ALLOC_PATH_MAX'
  360 |     VSTR_FIXED(path, MICROPY_ALLOC_PATH_MAX)
      |                      ^~~~~~~~~~~~~~~~~~~~~~
../../lib/micropython/py/builtinimport.c:360:16: error: unused variable 'path_buf' [-Werror=unused-variable]
  360 |     VSTR_FIXED(path, MICROPY_ALLOC_PATH_MAX)
      |                ^~~~
../../lib/micropython/py/misc.h:169:51: note: in definition of macro 'VSTR_FIXED'
  169 |  VSTR_FIXED(vstr, alloc) vstr_t vstr; char vstr##_buf[(alloc)]; vstr_init_fixed_buf(&vstr, (alloc), vstr##_buf);
      |                                            ^~~~

cc1: all warnings being treated as errors
make[1]: * [../../lib/micropython/py/mkrules.mk:63 : build/py/builtinimport.o] Erreur 1
make[1] : on quitte le répertoire « /home/leomlr/Documents/micropython-microbit-v2/src/codal_port »
```

mytechnotalent, 2020-10-25:
@leomlr I want to confirm you are doing these exact steps.  Please first before anything remove the build directory or `make clean` then do the following.

1. Go into root of repo and `git submodule update --init`
2. 'git clone https://github.com/lancaster-university/codal.git lib/codal'
    'git clone --recurse-submodules https://github.com/lancaster-university/codal-core.git lib/codal/libraries/codal-core'
    'git clone --recurse-submodules https://github.com/lancaster-university/codal-nrf52.git lib/codal/libraries/codal-nrf52'
    'git clone --recurse-submodules https://github.com/microbit-foundation/codal-microbit-nrf5sdk.git lib/codal/libraries/codal-microbit-nrf5sdk'
    'git clone --recurse-submodules https://github.com/lancaster-university/codal-microbit-v2.git lib/codal/libraries/codal-microbit-v2'
3. `cd src`
    `make`

I have successfully built so it is working. Can you try this again?

microbit-carlos, 2020-11-25:
@leomlr the build and build instructions has changed a bit since you've open this issue. Could you have another attempt to see if it is working for you?

microbit-carlos, 2020-11-30:
@aivarannamaa also had this issue in Ubuntu 20.04:

> I followed the instructions in README.md. When doing `make` in `src`, I got following errors:
> 
> ```
> In file included from ../../lib/micropython/py/qstr.h:30,
>                  from ../../lib/micropython/py/lexer.h:32,
>                  from ../../lib/micropython/py/compile.h:29,
>                  from ../../lib/micropython/py/builtinimport.c:32:
> ../../lib/micropython/py/builtinimport.c: In function 'mp_builtin___import__':
> ./mpconfigport.h:35:50: error: 'PATH_MAX' undeclared (first use in this function)
>    35 | #define MICROPY_ALLOC_PATH_MAX                  (PATH_MAX)
>       |                                                  ^~~~~~~~
> ../../lib/micropython/py/misc.h:169:63: note: in definition of macro 'VSTR_FIXED'
>   169 | define VSTR_FIXED(vstr, alloc) vstr_t vstr; char vstr##_buf[(alloc)]; vstr_init_fixed_buf(&vstr, (alloc), vstr##_buf);
>       |                                                              ^~~~~
> 
> ../../lib/micropython/py/builtinimport.c:360:22: note: in expansion of macro 'MICROPY_ALLOC_PATH_MAX'
>   360 |     VSTR_FIXED(path, MICROPY_ALLOC_PATH_MAX)
>       |                      ^~~~~~~~~~~~~~~~~~~~~~
> ./mpconfigport.h:35:50: note: each undeclared identifier is reported only once for each function it appears in
>    35 | #define MICROPY_ALLOC_PATH_MAX                  (PATH_MAX)
>       |                                                  ^~~~~~~~
> ../../lib/micropython/py/misc.h:169:63: note: in definition of macro 'VSTR_FIXED'
>   169 | define VSTR_FIXED(vstr, alloc) vstr_t vstr; char vstr##_buf[(alloc)]; vstr_init_fixed_buf(&vstr, (alloc), vstr##_buf);
>       |                                                              ^~~~~
> 
> ../../lib/micropython/py/builtinimport.c:360:22: note: in expansion of macro 'MICROPY_ALLOC_PATH_MAX'
>   360 |     VSTR_FIXED(path, MICROPY_ALLOC_PATH_MAX)
>       |                      ^~~~~~~~~~~~~~~~~~~~~~
> ../../lib/micropython/py/builtinimport.c:360:16: error: unused variable 'path_buf' [-Werror=unused-variable]
>   360 |     VSTR_FIXED(path, MICROPY_ALLOC_PATH_MAX)
>       |                ^~~~
> ../../lib/micropython/py/misc.h:169:51: note: in definition of macro 'VSTR_FIXED'
>   169 | #define VSTR_FIXED(vstr, alloc) vstr_t vstr; char vstr##_buf[(alloc)]; vstr_init_fixed_buf(&vstr, (alloc), vstr##_buf);
>       |                                                   ^~~~
> cc1: all warnings being treated as errors
> make[1]: *** [../../lib/micropython/py/mkrules.mk:63: build/py/builtinimport.o] Error 1
> make[1]: Leaving directory '/home/annamaa/python_stuff/mibi2/src/codal_port'
> make: *** [Makefile:44: libmicropython] Error 2
> ```

What version of GCC are you using? Any other system information would be quite useful as well.

mytechnotalent, 2020-11-30:
@aivarannamaa did you `git pull` and remove the build directory before trying to rebuild?  You might have already completed these steps but wanted to see if this might have been missed.

microbit-carlos, 2020-11-30:
I can confirm the same error in Ubuntu 18.04 with the apt version of arm gcc :
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I then downloaded and used gcc arm v7.3.1 and the build finished successfully:

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
$ cmake --version
cmake version 3.10.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

-----

However in Ubuntu 20.04 I get the same issue with both the apt v9.2.1, and the manually downloaded v7.3.1:

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
```
$ cmake --version
cmake version 3.16.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

mytechnotalent, 2020-11-30:
@microbit-carlos gcc arm v7.3.1 on both Ubuntu 18.04 and Ubuntu 20.04 is the solution?

microbit-carlos, 2020-11-30:
Well, kind of.
Based on my previous message it didn't work in Ubuntu 20.04 with arm gcc v7.3.1 (neither version worked).

However I think that's because CMake was still calling v9.2.1.
After uninstalling the `gcc-arm-none-eabi` package and setting up the environment to build with the manually downloaded v7.3.1, then it worked in 20.04.

The interesting part is that both debian packages didn't work,  v6.3.1 in 18.04 and v9.2.1 in 20.04, but the downloaded v7 did. So it's likely not an issue with the GCC version but some difference with the debian package?

mytechnotalent, 2020-11-30:
> However I think that's because CMake was still calling v9.2.1.
> After uninstalling the `gcc-arm-none-eabi` package and setting up the environment to build with the manually downloaded v7.3.1, then it worked in 20.04.

That is what I meant to say but it did not come out properly :)  

I think the issue is after you uninstalled `gcc-arm-none-eabi` and then setting up the new env with the v7.3.1 there is some dependency that was failing silently somewhere however when you manually uninstalled and reinstalled it grabbed the proper version of the dependency.

I would suggest we make a note in the README.md that if you are having issues very explicitly do the steps to ensure no side effects.

microbit-carlos, 2020-11-30:
Installing v7.3.1 via the `team-gcc-arm-embedded` PPA and  `gcc-arm-embedded` package (https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa) also works on Ubuntu 18.04.

mytechnotalent, 2020-11-30:
Sounds like we have a consistent and reproducible workaround.

jaustin, 2020-12-14:
@dpgeorge is going to set PATH_MAX in future builds which should resolve this inconsistency

dpgeorge, 2020-12-14:
Hopefully this is fixed by 47ebc73a9098ec24c758733bf1a77f22233aef93

microbit-carlos, 2020-12-14:
Thanks Damien!

I had a quick look around I think the macro was not found due to a missing include `#include <sys/syslimits.h>`, although I am not sure why this is automatically resolved with the Arm-packaged compiler, and not the Debian-packaged one.
And of course, as you had already confirmed, the value is too large anyway (1024), so manually setting a lower value in this port is definitely the right call. Anyway, I though it would be worth mentioning this in case the upstream unix port might need the sys include as well, or maybe mpy-cross might need it?

Gave this a spin in an Ubuntu 20.04 VM with the default `gcc-arm-none-eabi` package and can confirm the fix! 👍

dpgeorge, 2020-12-15:
> Anyway, I though it would be worth mentioning this in case the upstream unix port might need the sys include as well, or maybe mpy-cross might need it?

Ok, thanks.  I can't recall any upstream report of PATH_MAX being undefined, but if we get one I'll know what to do!

closed 2020-12-15 by microbit-carlos