microbit-foundation/micropython-microbit-v2 #6: Move to S113 bluetooth port

jaustin, 2020-10-28:
Since https://github.com/lancaster-university/codal-microbit-v2/commit/a4f0d11558ae195076fb6fe6b291af8bf83733c2 we have S113 in CODAL and a smaller bootloader, leading to more flash free for the MicroPython filesystem.

@dpgeorge can you please pull in the DAL update and check everything ends up in appropriate flash locations?

jaustin, 2020-10-30:
@dpgeorge a headsup that @finneyj has also since updated the virtual pin volume behaviour so you don't have to keep your own *4 in there too.

dpgeorge, 2020-11-02:
The latest CODAL with S113 seems to work, in terms of flash layout.  But there were other changes to the CODAL which have broken sound a little bit.  Working on it now.

> updated the virtual pin volume behaviour so you don't have to keep your own *4 in there too.

There is no *4 in there... the `microbit.set_volume(v)` function takes a value 0-255 and I think that API to the user should stay as 0-255.

jaustin, 2020-11-13:
Yea, sorry, on the volumes I thought you were manipulating virtual pin volume not mixer volume, so ignore the *4 comment and continue as you expected :)

closed 2020-11-13 by jaustin