microbit-foundation/micropython-microbit-v2 #9: Read light level returns 0

microbit-mark, 2020-11-05:
Originally raised by @crouse12 in https://github.com/bbcmicrobit/PythonEditor/issues/338

``` Python
from microbit import *
while True:
    lightLevel = display.read_light_level()
    display.scroll(lightLevel)
    sleep(1000)
```
Returns 0 consistently on the display.

microbit v2.

Unique ID: 9904360258434e45004a400e0000005f000000009796990b
HIC ID: 9796990b
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Incompatible image detection: 1
Page erasing: 0
Daplink Mode: Interface
Interface Version: 0254
Bootloader Version: 0254
Git SHA: ec3fec91e815b1fe27cefb8bc4ffa85ca3317502
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0x23cc2063
Interface CRC: 0xd0f0bacf
Remount count: 0
URL: https://microbit.org/device/?id=9904&v=0254

python editor
https://python.microbit.org/v/beta

OS: window 10
Chrome version: 86.0.4240.111

---

I have tried to reproduce and can't using either V1 or v2 board in python beta editor

[image]

microbit-mark, 2020-11-05:
@crouse12. Do you see the same MicroPython version when you click on Help in the editor?
[image]

microbit-carlos, 2020-11-05:
As this should be a fairly standard CODAL pass-through and I happened to have a CODAL project open in the background and gave this a quick test and it doesn't work in that layer: https://github.com/lancaster-university/codal-microbit-v2/issues/14

crouse12, 2020-11-05:
2.1.0-beta.5 and 2.1.0-beta.6 have the same problem.

> @crouse12. Do you see the same MicroPython version when you click on Help in the editor?
> [image]

microbit-carlos, 2020-11-09:
This has been fixed in CODAL, so it should be available in the next CI builds or the next Python Editor release.
We'll close here once we can confirm the original example script works.

microbit-carlos, 2020-11-23:
@microbit-mark @martinwork this can be tested already in https://python.microbit.org/v/beta (beta.7)

martinwork, 2020-11-23:
Testing code below in:
Python Editor Version: 2.1.0-beta.7 MicroPython Version: 1.0.1/2.0-44a75bb
lancaster-university/codal-microbit-v2, branch master
lancaster-university/microbit#v2.2.0-rc6
MakeCode beta

With CPP: 
V1 shows 255 first, followed by values that change according to light level.
V2 shows stable readings from reset, power up and download

With Python:
V1 shows a high first reading, then good readings
V2 shows a zero reading after pressing reset, but not after power up or download

With MakeCode beta:
V1 shows 255, then good readings
V2 shows good readings

```
from microbit import *
while True:
    lightLevel = display.read_light_level()
    display.scroll(lightLevel)
    sleep(1000)

```

```
// This tests the LEDs light level

#include "MicroBit.h"

MicroBit uBit;

void forever()
{
  while (true)
  {
    ManagedString s( uBit.display.readLightLevel());
    uBit.display.scroll( s);
    uBit.sleep( 0);
  }
}

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    create_fiber( forever);

    // If main exits, there may still be other fibers running or registered event handlers etc.
    // Simply release this fiber, which will mean we enter the scheduler. Worse case, we then
    // sit in the idle task forever, in a power efficient sleep.
    release_fiber();
}

```
```

basic.forever(function () {
    basic.showNumber(input.lightLevel())
})

```

dpgeorge, 2020-11-24:
> With Python:
> V2 shows a zero reading after pressing reset, but not after power up or download

Testing on the lastest master at 6565f16878b8b472c3b020f3bf0d8b3bf1a0a124, after pressing the reset button `display.read_light_level()` returns a non-zero value for me (as long as there is a bright enough light on the display).

microbit-carlos, 2020-11-25:
> With CPP:
> V1 shows 255 first, followed by values that change according to light level.
> V2 shows stable readings from reset, power up and download

This is now expected as there is an open issue in the DAL repo and this is the current CODAL implementation.

> With Python:
> V1 shows a high first reading, then good readings
> V2 shows a zero reading after pressing reset, but not after power up or download

With this programme in V2:
```python
from microbit import *
while True:
    lightLevel = display.read_light_level()
    print(lightLevel)
    display.scroll(lightLevel)
    sleep(1000)
```

I'm getting non-zero values after reset (with the phone flash light).

Could you try again Martin? If it works with the MicroPython version in https://python.microbit.org/v/2.1 we can close this issue.

martinwork, 2020-11-25:
V2 is OK. V1 starts with a high reading.

microbit-carlos, 2020-11-25:
Thanks Martin!
The DAL behaviour is captured in https://github.com/lancaster-university/microbit-dal/issues/426, and MicroPython for V2 works as expected with CODAL, so we can close this issue.

closed 2020-11-25 by microbit-carlos

crouse12, 2020-11-25:
Return 0 when I use microbit v2

[image]

microbit-carlos, 2020-11-25:
Make sure you have a good light source on top of it, it will likely read 0 in normal ambient light.