microbit-foundation/micropython-microbit-v2 #11: Can't generate hex file

mytechnotalent, 2020-11-08:
When running `make` I am not able to generate a hex binary.

`
.o
[ 90%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/bluetooth/MicroBitBLEServices.cpp.o
[ 91%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/bluetooth/MicroBitBLETypes.cpp.o
[ 91%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/bluetooth/MicroBitButtonService.cpp.o
[ 92%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/bluetooth/MicroBitEddystone.cpp.o
[ 92%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/bluetooth/MicroBitEventService.cpp.o
[ 93%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/bluetooth/MicroBitIOPinService.cpp.o
[ 93%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/bluetooth/MicroBitLEDService.cpp.o
[ 94%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/bluetooth/MicroBitMagnetometerService.cpp.o
[ 94%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/bluetooth/MicroBitPartialFlashingService.cpp.o
[ 95%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/bluetooth/MicroBitTemperatureService.cpp.o
[ 95%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/bluetooth/MicroBitUARTService.cpp.o
[ 96%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/compat/MbedMemberFunctionCallback.cpp.o
[ 96%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/model/MicroBit.cpp.o
[ 97%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/model/MicroBitIO.cpp.o
[ 97%] Linking CXX static library ../../libcodal-microbit-v2.a
[ 97%] Built target codal-microbit-v2
Scanning dependencies of target MICROBIT
[ 97%] Building CXX object CMakeFiles/MICROBIT.dir/Users/developer/Documents/micropython-microbit-v2/src/codal_app/mphalport.cpp.o
[ 97%] Building CXX object CMakeFiles/MICROBIT.dir/Users/developer/Documents/micropython-microbit-v2/src/codal_app/main.cpp.o
[ 98%] Building CXX object CMakeFiles/MICROBIT.dir/Users/developer/Documents/micropython-microbit-v2/src/codal_app/microbithal_microphone.cpp.o
[ 98%] Building CXX object CMakeFiles/MICROBIT.dir/Users/developer/Documents/micropython-microbit-v2/src/codal_app/microbithal.cpp.o
[ 99%] Linking CXX executable MICROBIT
[ 99%] Built target MICROBIT
Scanning dependencies of target MICROBIT_hex
Scanning dependencies of target MICROBIT_bin
[ 99%] converting to bin file.
[100%] converting to hex file.
[100%] Built target MICROBIT_bin
[100%] Built target MICROBIT_hex
size ../lib/codal/build/MICROBIT
__TEXT  __DATA  __OBJC  others  dec     hex
error: /Library/Developer/CommandLineTools/usr/bin/size: file: ../lib/codal/build/MICROBIT is not an object file
make: *** [all] Error 1
developer@mac-developer src % 
`

mytechnotalent, 2020-11-08:
@microbit-carlos it does generate the hex file after all in `lib/codal/build` but originally it was putting the .hex in the /src dir so I still think you have a minor issue here.

mytechnotalent, 2020-11-08:
__OBJC others dec hex error: /Library/Developer/CommandLineTools/usr/bin/size: file: ../lib/codal/build/MICROBIT is not an object file make: *** [all] Error 1 developer@mac-developer src %

microbit-carlos, 2020-11-08:
Right, so this line in the makefile is failing because the hex file hasn't been moved to the root directory.
Is this doing a build from a clean clone of the repository, or does it contain any modifications?

mytechnotalent, 2020-11-08:
It has mods.  Here is a REPO detailing the mods as I created a tutorial for C Developers to help mod their own private firmware outside of upstream. https://github.com/mytechnotalent/MicroPython-micro-bit_MOD_C_Library

mytechnotalent, 2020-11-08:
This was a fresh clone of the repo however the mods were done after the fresh clone.

microbit-carlos, 2020-11-08:
Right, so the build works without the mods but fails with the mods?
If you haven't tried that yet, could you try building without the mods?

mytechnotalent, 2020-11-08:
I will try now I have not tried without the mods standby...

mytechnotalent, 2020-11-08:
I have deleted the cloned repo and re-cloned fresh and ran all the steps and built the firmware.  The result is the same.  Same error there is an error in main that is not copying the .hex over properly.  This has nothing to do with my custom mod which is good so we know we are consistent here.

microbit-carlos, 2020-11-08:
I just tried in a new clone, following the instructions in the README and got a working build in macOS without any issues.

Could you copy the full output of a build (after a `make clean`) from beginning to end?
It'd be really useful to also get the version of Arm GCC and CMake.

It sounds like your build is producing a lib/codal/build/MICROBIT file, but it is not an object file? Is your build generating a lib/codal/MICROBIT.hex and MICROBIT.bin file as well?

mytechnotalent, 2020-11-08:
@microbit-carlos 
```
developer@mac-developer Documents % git clone git@github.com:microbit-foundation/micropython-microbit-v2.git
Cloning into 'micropython-microbit-v2'...
remote: Enumerating objects: 568, done.
remote: Counting objects: 100% (568/568), done.
remote: Compressing objects: 100% (191/191), done.
remote: Total 568 (delta 384), reused 549 (delta 365), pack-reused 0
Receiving objects: 100% (568/568), 168.71 KiB | 177.00 KiB/s, done.
Resolving deltas: 100% (384/384), done.
developer@mac-developer Documents % ls
circuitpython		micropython
esptool			micropython-microbit-v2
developer@mac-developer Documents % cd micropython-microbit-v2 
developer@mac-developer micropython-microbit-v2 % git submodule update --init
Submodule 'lib/micropython' (https://github.com/micropython/micropython) registered for path 'lib/micropython'
Cloning into '/Users/developer/Documents/micropython-microbit-v2/lib/micropython'...
Submodule path 'lib/micropython': checked out 'e54626f4c1b5cb4de0b5dc86d820bb36c1d6499f'
developer@mac-developer micropython-microbit-v2 % git clone https://github.com/lancaster-university/codal.git lib/codal
Cloning into 'lib/codal'...
remote: Enumerating objects: 8, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 2913 (delta 1), reused 3 (delta 1), pack-reused 2905
Receiving objects: 100% (2913/2913), 1.36 MiB | 103.00 KiB/s, done.
Resolving deltas: 100% (1916/1916), done.
developer@mac-developer micropython-microbit-v2 % git clone --recurse-submodules https://github.com/lancaster-university/codal-core.git lib/codal/libraries/codal-core
Cloning into 'lib/codal/libraries/codal-core'...
remote: Enumerating objects: 31, done.
remote: Counting objects: 100% (31/31), done.
remote: Compressing objects: 100% (22/22), done.
remote: Total 7299 (delta 12), reused 18 (delta 9), pack-reused 7268
Receiving objects: 100% (7299/7299), 1.49 MiB | 257.00 KiB/s, done.
Resolving deltas: 100% (5547/5547), done.
developer@mac-developer micropython-microbit-v2 % git clone --recurse-submodules https://github.com/lancaster-university/codal-nrf52.git lib/codal/libraries/codal-nrf52
Cloning into 'lib/codal/libraries/codal-nrf52'...
remote: Enumerating objects: 175, done.
remote: Counting objects: 100% (175/175), done.
remote: Compressing objects: 100% (93/93), done.
remote: Total 895 (delta 121), reused 131 (delta 81), pack-reused 720
Receiving objects: 100% (895/895), 589.94 KiB | 574.00 KiB/s, done.
Resolving deltas: 100% (566/566), done.
Submodule 'nrfx' (https://github.com/NordicSemiconductor/nrfx) registered for path 'nrfx'
Cloning into '/Users/developer/Documents/micropython-microbit-v2/lib/codal/libraries/codal-nrf52/nrfx'...
remote: Enumerating objects: 155, done.        
remote: Counting objects: 100% (155/155), done.        
remote: Compressing objects: 100% (123/123), done.        
remote: Total 2670 (delta 88), reused 45 (delta 32), pack-reused 2515        
Receiving objects: 100% (2670/2670), 4.27 MiB | 234.00 KiB/s, done.
Resolving deltas: 100% (2150/2150), done.
Submodule path 'nrfx': checked out '9136253e271f978a8da735864fc50d60e0ed68c6'
developer@mac-developer micropython-microbit-v2 % git clone --recurse-submodules https://github.com/microbit-foundation/codal-microbit-nrf5sdk.git lib/codal/libraries/codal-microbit-nrf5sdk
Cloning into 'lib/codal/libraries/codal-microbit-nrf5sdk'...
remote: Enumerating objects: 1626, done.
remote: Counting objects: 100% (1626/1626), done.
remote: Compressing objects: 100% (929/929), done.
remote: Total 1626 (delta 689), reused 1578 (delta 656), pack-reused 0
Receiving objects: 100% (1626/1626), 19.89 MiB | 375.00 KiB/s, done.
Resolving deltas: 100% (689/689), done.
developer@mac-developer micropython-microbit-v2 % git clone --recurse-submodules https://github.com/lancaster-university/codal-microbit-v2.git lib/codal/libraries/codal-microbit-v2
Cloning into 'lib/codal/libraries/codal-microbit-v2'...
remote: Enumerating objects: 33, done.
remote: Counting objects: 100% (33/33), done.
remote: Compressing objects: 100% (26/26), done.
remote: Total 1599 (delta 12), reused 17 (delta 7), pack-reused 1566
Receiving objects: 100% (1599/1599), 910.88 KiB | 322.00 KiB/s, done.
Resolving deltas: 100% (1108/1108), done.
developer@mac-developer micropython-microbit-v2 % cd src 
developer@mac-developer src % make
/Library/Developer/CommandLineTools/usr/bin/make -C codal_port
[... 484 lines cut ...]
[ 96%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/compat/MbedMemberFunctionCallback.cpp.o
[ 96%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/model/MicroBit.cpp.o
[ 97%] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/model/MicroBitIO.cpp.o
[ 97%] Linking CXX static library ../../libcodal-microbit-v2.a
[ 97%] Built target codal-microbit-v2
Scanning dependencies of target MICROBIT
[ 97%] Building CXX object CMakeFiles/MICROBIT.dir/Users/developer/Documents/micropython-microbit-v2/src/codal_app/microbithal.cpp.o
[ 97%] Building CXX object CMakeFiles/MICROBIT.dir/Users/developer/Documents/micropython-microbit-v2/src/codal_app/main.cpp.o
[ 98%] Building CXX object CMakeFiles/MICROBIT.dir/Users/developer/Documents/micropython-microbit-v2/src/codal_app/microbithal_audio.cpp.o
[... cut ...]
[ 99%] Linking CXX executable MICROBIT
[ 99%] Built target MICROBIT
Scanning dependencies of target MICROBIT_hex
Scanning dependencies of target MICROBIT_bin
[100%] converting to hex file.
[100%] converting to bin file.
[100%] Built target MICROBIT_bin
[100%] Built target MICROBIT_hex
size ../lib/codal/build/MICROBIT
__TEXT	__DATA	__OBJC	others	dec	hex
error: /Library/Developer/CommandLineTools/usr/bin/size: file: ../lib/codal/build/MICROBIT is not an object file
make: *** [all] Error 1
developer@mac-developer src % 
```

mytechnotalent, 2020-11-08:
> I just tried in a new clone, following the instructions in the README and got a working build in macOS without any issues.
>
> Could you copy the full output of a build (after a `make clean`) from [...]

It is creating a `lib/codal/MICROBIT.hex` and a `lib/codal/MICROBIT.bin` yes.

microbit-carlos, 2020-11-08:
Right, output looks good and sounds like it's generating everything. It's possible `size` in your environment is executing a tool from a compiler incompatible with this kind of object file.

Can you try `size --version`?
In the meantime, if ` arm-none-eabi-size` is in your path you can also edit that line in the makefile to run that instead of just `size` (or just comment out that line, is there just for info).

mytechnotalent, 2020-11-08:
```
developer@mac-developer src % size --version
error: /Library/Developer/CommandLineTools/usr/bin/size: invalid argument --
Usage: /Library/Developer/CommandLineTools/usr/bin/size [-m] [-l] [-x] [--] [[-arch <arch_flag>] ...] [file ...]
developer@mac-developer src % 
```

Does not give a version number.

I do not see `arm-none-eabi-size` in either `makefile` as `arm-none-eabi-size` is in my path as you can see:

```
developer@mac-developer src % arm-none-eabi-size
arm-none-eabi-size: 'a.out': No such file
developer@mac-developer src % 
```

mytechnotalent, 2020-11-08:
```
developer@mac-developer src % arm-none-eabi-size --version
GNU size (GNU Arm Embedded Toolchain 9-2020-q2-update) 2.34.0.20200428
Copyright (C) 2020 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.
developer@mac-developer src % 
```

microbit-carlos, 2020-11-08:
Right, looking at the output from `size --version` I am not sure what programme `size` is in your environment, but that's definitely the issue.

> I do not see `arm-none-eabi-size` in either `makefile` as `arm-none-eabi-size` is in my path as you can see:

What I meant is that you can edit this line to replace `size` with `arm-none-eabi-size`, or comment it out:
https://github.com/microbit-foundation/micropython-microbit-v2/blob/3e09245a46799711b8a285f0a2349e5c13d6b948/src/Makefile#L11

@dpgeorge I'm not too familiar with the different versions of `size`, should we make this change to the Makefile?

mytechnotalent, 2020-11-08:
@microbit-carlos @dpgeorge I have successfully tested the change on line 11:

micropython-microbit-v2/src/Makefile

Line 11 in 3e09245
 arm-none-eabi-size $(CODAL_DIR)/build/MICROBIT 

It works from a fresh clone and with my modified C libs.  If this does not break on your builds I would suggest making this a perm change for everyone as you might be suggesting @microbit-carlos.

dpgeorge, 2020-11-09:
Yes this should be `arm-none-eabi-size`.  Fixed in f98d7b1cc835718cd0d2a75e822aa5eafe594961

closed 2020-11-09 by dpgeorge