microbit-foundation/micropython-microbit-v2 #12: Lcd.off and Lcd.on problem

crouse12, 2020-11-09:
Recently, I try to create scratch blocks for microbit v2.
I found that python editor beta 2.1.0-beta.6 for microbit v2 has a bug for led screen.
When I use the following python program, led screen on microbit v2 has some errors.
Can anyone help me?
PS. python editor Editor Version: 2.0.1 for microbit v1 does not have this bug

[image]

dpgeorge, 2020-11-09:
Thanks for the report.  The issue is that pin10 is connected to the display and `display.on()` does not fully reset the display, so the `pin10.write_analog(0)` has a side effect of leaving the last column all lit up.

@microbit-carlos would we expect `uBit.display.enable()` to fully reset the display?

microbit-carlos, 2020-11-09:
Yes, I think it should: https://github.com/lancaster-university/codal-microbit-v2/issues/21

pin10 is also a display pin on V1, and this doesn't seem to be an issue in V1.

dpgeorge, 2020-11-17:
I tested this with the latest build using the latest CODAL and it is fixed.

closed 2020-11-17 by dpgeorge