microbit-foundation/micropython-microbit-v2 #13: Help anchor for Connect scrolls too low

martinwork, 2020-11-09:
https://python-editor-2-1-0-beta-6.microbit.org/help.html#command-connect scrolls to half way down the Connect panel, so Connect isn't visible.

microbit-carlos, 2020-11-09:
Good catch Martin, let's move this to https://github.com/bbcmicrobit/PythonEditor as that's were we develop the editor: https://github.com/bbcmicrobit/PythonEditor/issues/341

closed 2020-11-09 by microbit-carlos

martinwork, 2020-11-09:
@microbit-carlos Is that were I should post any issues arising from the Foundation tests?

microbit-carlos, 2020-11-10:
Depending on the tests, the Python Editor checklist would cover things for the Python Editor rather than MicroPython and issue would probably go in https://github.com/bbcmicrobit/PythonEditor.

Any MicroPython functional tests from the microbit-test repository would cover things from MicroPython and should be raised in https://github.com/microbit-foundation/micropython-microbit-v2.