microbit-foundation/micropython-microbit-v2 #14: Add set and get touch modes for touch pins

microbit-carlos, 2020-11-12:
To be able to set and get the touch model for the logo and touch pins, between capacitive and resistive.

The default modes should be (I believe this part is implemented already):
- Logo touch will be default to capacitive
- Edge connector pins will be resistive

jaustin, 2020-11-13:
The uBit object already constructs these with the appropriate defaults, but there should be a way to switch them over to the other modes. This is done in CODAL bu calling isTouched() with an argument to determine the mode

https://github.com/lancaster-university/codal-nrf52/blob/master/source/NRF52Pin.cpp#L520

Here's usage in MakeCode...
https://github.com/microsoft/pxt-microbit/blob/master/libs/core/touchmode.cpp#L40

Pins that can have their touchmode changed
* logo
* P0
* P1
* P2

(this list is defined by the hardware because these pins have 10M pullups that make the resistive mode work)

`logo` is it's own class with only `is_touched()' so that needs extending, as do pin0,1,2

Proposed Python API:
```
pin0.set_touch_mode(pin0.CAPACITIVE)
pin_logo.set_touch_mode(pin_logo.RESISTIVE)
```

dpgeorge, 2020-12-21:
`pin.set_touch_mode(mode)` method and corresponding constants were added in cbc3cfccb7a00ea40d377f2a49628625b32efdf8

microbit-carlos, 2020-12-21:
Gave this a test and all looks good, thanks Damien!

closed 2020-12-21 by microbit-carlos