microbit-foundation/micropython-microbit-v2 #15: Serial "delayed" characters on serial, specially when playing sound

microbit-carlos, 2020-11-12:
This should be fixed with the latest CODAL, but we still need to test it and confirm it's the case and if there are still issues report back to CODAL.

When using sound and the REPL sometimes it get's into a state where typing a character only shows back 2 or 3 keystrokes in the future.

jaustin, 2020-11-13:
If this isn't fixed it could be a blocker for Mu so testing with latest DAL (V0.2.19) is high priority. @martinwork

martinwork, 2020-11-23:
The REPL seems OK now.

Tested with Editor Version: 2.1.0-beta.7 MicroPython Version: 1.0.1/2.0-44a75bb, typing and using the REPL history with these commands:
music.play(music.BLUES, wait=False)
music.pitch(440)
music.stop()

I noticed that music.play cancels any previous music.play or music.pitch, but music.pitch doesn't cancel a previous music.play. The music.pitch sounds briefly, and the blues keep playing. Is this expected?

microbit-carlos, 2020-11-25:
Thanks Martin, I've opened https://github.com/microbit-foundation/micropython-microbit-v2/issues/28 to discuss the music question.

About the UART I was able to trigger it by pasting code into the REPL quickly until some characters got lost.

In this GIF (>3MBs, so it might take a minute lo load) I paste `audio.play(Sound.HAPPY)` several times at a moderate rate, which doesn't trigger the issue. I then try again pasting quite fast and it triggers once you start seeing the exceptions printed.
At that point I only press enter and you can see the previous characters appear:

[image]

microbit-carlos, 2020-11-25:
I wasn't able to replicate this with something like `print('hello world')`, so it might be related to the audio stuff, or to something that uses a lot of CPU.

martinwork, 2020-11-25:
It seems better than it was. I have only been testing in the web app. 

Could there be a buffering problem that isn't strictly "serial"? i.e. not CODAL serial

Trying in tera term, using up-arrow/return to repeat a single command. There seems to be a difference between these:
>>> music.play(music.BLUES,wait=False)
>>> audio.play(Sound.HAPPY)

Pressing up-arrow/return as fast as I can, after about 10-15 audio.plays have executed:
>>> audio.play(Sound.HAPPY)
>>> audio.play(Sound.HAPPY)
>>> audio.play(Sound.HAPPY)
>>> [AA
... [AA
... [AA
... [AA

I eventually got the same result with a V1 and microbit.display.show("Hello")

I can paste many lines in one go that show a single character (multiple repeats of the alphabet), but 2 characters locks up easily -  more so on V1

microbit.display.show("A0")
microbit.display.show("B0")
microbit.display.show("C0")
microbit.display.show("D0")
microbit.display.show("E0")
microbit.display.show("F0")
microbit.display.show("G0")
microbit.display.show("H0")
microbit.display.show("I0")
microbit.display.show("J0")
microbit.display.show("K0")
microbit.display.show("l0")
microbit.display.show("M0")
microbit.display.show("N0")
microbit.display.show("O0")
microbit.display.show("P0")
microbit.display.show("Q0")
microbit.display.show("R0")
microbit.display.show("S0")
microbit.display.show("T0")
microbit.display.show("U0")
microbit.display.show("V0")
microbit.display.show("W0")
microbit.display.show("X0")
microbit.display.show("Y0")
microbit.display.show("Z0")

microbit.display.show("A")
microbit.display.show("B")
microbit.display.show("C")
microbit.display.show("D")
microbit.display.show("E")
microbit.display.show("F")
microbit.display.show("G")
microbit.display.show("H")
microbit.display.show("I")
microbit.display.show("J")
microbit.display.show("K")
microbit.display.show("L")
microbit.display.show("M")
microbit.display.show("N")
microbit.display.show("O")
microbit.display.show("P")
microbit.display.show("Q")
microbit.display.show("R")
microbit.display.show("S")
microbit.display.show("T")
microbit.display.show("U")
microbit.display.show("V")
microbit.display.show("W")
microbit.display.show("X")
microbit.display.show("Y")
microbit.display.show("Z")

jaustin, 2020-12-14:
@microbit-carlos to post the gif and Martin could you please re-test once #35 is resolved in case these are related?

martinwork, 2020-12-14:
@jaustin  Will do!

microbit-carlos, 2020-12-14:
The GIF was actually higher up in this thread (it's a bit heavy, so maybe it didn't load when we were looking at this page):
https://github.com/microbit-foundation/micropython-microbit-v2/issues/15#issuecomment-733776145

[image]

I've tried this with the latest hex from the CI (https://github.com/microbit-foundation/micropython-microbit-v2/actions/runs/420785120) and can still replicate by copy/pasting `audio.play(Sound.HAPPY)` as quickly as I can as shown in the second half of this GIF.

martinwork, 2020-12-14:
Pasting into tera term shows the difference between V1 and V2.

The results are the same if I paste the whole thing  or type "import microbit" and paste the rest.

It looks to be about filling up the RX buffer, because executing the commands is delaying throughput.

After the output below, pressing SPACE is echoed OK by V1 but V2 echoes with previous sent characters.

The V2 response is like https://github.com/lancaster-university/codal-microbit-v2/issues/46.

Pasting this into tera term
```
import microbit
microbit.display.show("Aa")
microbit.display.show("Bb")
microbit.display.show("Cc")
microbit.display.show("Dd")
microbit.display.show("Ee")
microbit.display.show("Ff")

```

V1 says
```
>>> import microbit
>>> microbit.display.show("Aa")
>>> microbit.display.show("Bb")
>>> microbit.display.show("Cc")
>>> microbi

```

MICROBIT-MICROPYTHON-967c0765* V2 says:
```
>>> import microbit
>>> microbit.display.show("Aa")
>>> show("Ff")
Traceback (most recent call last):
  File "<stdin>", in <module>
NameError: name 'show' isn't defined
>>> splay.show("Bb")
Traceback (most recent call last):
  File "<stdin>", in <module>
NameError: name 'splay' isn't defined
>>> microbit.display.show("Cc")
>>> microbit.display.show("Dd")
>>> microbit.display.show("Ee")
>>> microbit.display

```

microbit-carlos, 2020-12-21:
I can't replicate the issue anymore with the latest CODAL changes.
Although when pasting Matin's example not all the content ends up in the REPL, but it stops at that point and typing works as expected (before typing would show previous characters):

```
import microbit
microbit.display.show("Aa")
microbit.display.show("Bb")
microbit.display.show("Cc")
microbit.display.show("Dd")
microbit.display.show("Ee")
microbit.display.show("Ff")
```

```
>>> import microbit
>>> microbit.display.show("Aa")
>>> microbit.display.show("Bb")
>>> microbit.display.show("Cc")
>>> microbit.display.show("Dd")
>>> microbit.display.show("Ee")
>>> microbit.dis
```

Paste mode and raw repl seem to work fine, so I'll close this.

closed 2020-12-21 by microbit-carlos