microbit-foundation/micropython-microbit-v2 #16: Calibration persistance

microbit-carlos, 2020-11-12:
Calibration is already implemented, and the CODAL persistence is probably working, but we need to test it still.

jaustin, 2020-11-13:
Persistance was not working as of CODAL V0.2.18 but @finneyj is implementing it.
I don't think this needs action from anyone here at this stage but it must be tested as soon as we build with the first version of the DAL with fixed persistance. @martinwork can you please track that?

finneyj, 2020-11-13:
yes - just pending the off-chip-storage branch merge @martinwork, which also contains unified API wor that allows us to do on chip persistent storage of calibration data (among other things).

martinwork, 2020-11-13:
@jaustin @finney We have an issue open in the tests repo.

microbit-carlos, 2020-11-16:
Implemented in the latest CODAL release: https://github.com/lancaster-university/codal-microbit-v2/issues/12#issuecomment-728159042

martinwork, 2020-11-20:
Tested OK with micropython and makecode.

Calibration is retained after reset and power cycle.

It survives a WebUSB partial flash but not a full flash.

microbit-carlos, 2020-11-23:
Great, thanks Martin!

closed 2020-11-23 by microbit-carlos