microbit-foundation/micropython-microbit-v2 #17: Ability to Turn off built-in speaker

microbit-carlos, 2020-11-12:
We still need a way to turn off the built-in speaker and still have the sound coming out of pin0.

jaustin, 2020-11-13:
I thought this was implemented. The underlying DAL method is 
https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/MicroBitAudio.cpp#L116
And the API in MicroPython `microbit.pin_speaker.diable()` (though if @dpgeorge you have a strong preference for this not to be on the pin_speaker object because the operation is performed on the mixer, we can accommodate other optoins. However, this is an important thing to be able to do as soon as the device is available.

jaustin, 2020-11-16:
This depends on the behaviour we want for pin retargeting and pin_audio too.
We want the mixer to be transparent, so when a user specifies, say, P2 for their audio, we weill remap the mixer to P0 and play via the mixer.

This means, for now, there will be no direct access APIs to the speaker, so the disable method can safely call  MicroBitAudio.cpp#L116

So, we need a location for the speaker disable.

What happens if I do the following
```python
microbit.pin_speaker.disable()
music.play( "abc", pin=Pins.pin_speaker,) #Do not make a noise because speaker is disabled
microbit.pin_speaker.enable()
music.play( "abc", pin=Pins.pin_speaker,) #makes noise again
```

dpgeorge, 2020-11-19:
`pin_speaker.disable()` and `pin_speaker.enable()` have been implemented in 40e9bb687eb561cf590d151c6afa35efbcd4fec0

closed 2020-11-19 by dpgeorge