microbit-foundation/micropython-microbit-v2 #18: Prepare for beta releases

microbit-carlos, 2020-11-12:
Things we need to do to formalise beta releases:
- Use a tag version fo codal-microbit-v2
    - This repo is now tag and each tag conntains an updtaed targe-lock.json that points to specific commits from its dependencies
- Change the version string to v2.0.0-beta.1 so we can start upreving the beta version
- Ensure the upstream MicroPython version is correctly included
    - This might be already done, but I believe it was an issue before

jaustin, 2020-11-13:
This is essential so that at device availability there is a usable well-versioned hex file for consumption by third party editors like EduBlocks, Mu, FiriaLabs, VitaSciences, Thonny etc!

jaustin, 2020-11-16:
In order to take advantage of the CODAL tagging infrastructure, we need to use build.py to fetch the dependencies. This probably means changing the way we fetch the build.py script, as that currently comes from upstream codal. (lancaster-university/codal, not lancaster-university/microbit-v2-samples, which are both the same kind of thing).

Now that nothing is private, using a submodule for codal-v2-samples is reasonable but it would also be possible to use submodules for all the libraries but this is more work for updating.

In a normal codal project, you change the 'branch' to the latest tag. https://github.com/lancaster-university/microbit-v2-samples/blob/v0.2.11/codal.json#L5

@jamesadevine it would seem like for MicroPython it would be far easier if microbit-v2-samples also had tagged versions that specified the branch for codal-microbit-v2. Is there a reason not to do that, or an easy way to just feed the required branch to build.py?

jamesadevine, 2020-11-16:
Hi @jaustin,

I don't think there is a reason not to do that. You could use submodules to do that pretty easily, and maintainers could bump as required. An alternative would be to indeed pass the required branch to build.py, but that seems somewhat more error prone than submodules and would require some additional code to be written.

dpgeorge, 2020-11-24:
The following has now been done:
- CODAL made a submodule in 58b660a4e8e1d4616ebfa2485abfb4c31f216e27
- build.py no longer used, instead cmake is invoked directly and this automatically fetches the correct libraries pined at a given version, see 77961d8150bab24e1a85ae9d8a774dae3b88de8f
- correct versions for MP and micro:bit are implemented by 2f604f705fe5c0d810f965622eab9e1cb10b34e2
- version updated to v2.0.0-beta.1 in 9efa1fef849ff6eba6c0952e08d17d07c315431a

microbit-carlos, 2020-11-24:
Awesome, thanks Damien!

beta.1 has been tagged in https://github.com/microbit-foundation/micropython-microbit-v2/releases/tag/v2.0.0-beta.1

closed 2020-11-24 by microbit-carlos