microbit-foundation/micropython-microbit-v2 #20: Create SoundEvent class

microbit-carlos, 2020-11-13:
To capture the current 'loud' and 'quiet' options and give us the possibility to expand it in the future to other types of events.

Something along these lines:

```python
class MicroBitSoundEvent:

    LOUD = None
    QUIET = None

    def __init__(self, event_name: string) -> None:
        self.event_name = event_name

    def __str__(self):
        return "SoundEvent('{}')".format(self.event_name)

    def __repr__(self):
        return self.__str__()

MicroBitSoundEvent.LOUD = MicroBitSoundEvent("loud")
MicroBitSoundEvent.QUIET = MicroBitSoundEvent("quiet")
SoundEvent = MicroBitSoundEvent
```

So that the user code will end up:

```python
microphone.was_event(SoundEvent.LOUD)
```

jaustin, 2020-11-13:
@microbit-giles is currently drafting resources against this speculative API so important that the first public relese uses this.

dpgeorge, 2020-11-17:
Done in 18eb9de1e88afca0fb2615a2bc483d0e3a214fd3

closed 2020-11-17 by dpgeorge