microbit-foundation/micropython-microbit-v2 #24: music.pitch() plays only on pin0 after music.play(...pin=pin_speaker)

microbit-carlos, 2020-11-13:
Originally reported in https://github.com/bbcmicrobit/PythonEditor/issues/352 by @martinwork:

-----
With the code below, the first call to `music.pitch(440)` plays on both speaker and pin0.

But after `music.play(music.BLUES, pin=pin_speaker)`, the second call to `music.pitch(440)` plays only on pin0.

```python
from microbit import *
import music

while True:
    music.pitch(440)
    sleep(1000)
    music.stop()
    sleep(1000)
    music.play(music.BLUES, pin=pin_speaker)
    music.stop()

```

dpgeorge, 2020-11-19:
Should be fixed by e6e43f13a63415ea73f0637de410fbc44de4ce5f, but will need testing

martinwork, 2020-11-19:
How can I launch an editor with this change to test?
Will it arrive in https://python.microbit.org/v/beta?

microbit-carlos, 2020-11-19:
It will arrive to /v/beta tomorrow hopefully. In the meantime you can get hex files from the CI and test in the REPL or transfer a main.py using [Mu](https://tech.microbit.org/latest-revision/#python) (or [microFs](https://github.com/ntoll/microfs)).

https://github.com/microbit-foundation/micropython-microbit-v2/actions/runs/372190264

microbit-mark, 2020-11-19:
Setting `pin_speaker` no longer seems to disable `pin0` as it did previously. Is this related

```
import music
from microbit import *

music.play(music.BLUES, pin=pin_speaker)
```

plays both on speaker and pin0 in latest micropython, but in the editor it only plays on the speaker.

This works as expected. eg  only on pin0

```
import music
from microbit import *

music.play(music.BLUES, pin=pin0)
```

microbit-mark, 2020-11-19:
@martinwork This shows how to use a custom MicroPython version in [Mu](https://codewith.mu/) https://tech.microbit.org/software/beta/#mu

dpgeorge, 2020-11-19:
> This plays both on speaker and pin0 in latest micropython,

Ah, good catch, fixed by 44a75bb4d40390d6b809cfe4509ffccfa76eaf50

martinwork, 2020-11-20:
Thanks @microbit-carlos and @microbit-mark for the instructions.

I used the https://github.com/microbit-foundation/micropython-microbit-v2/commit/44a75bb4d40390d6b809cfe4509ffccfa76eaf50 build https://github.com/microbit-foundation/micropython-microbit-v2/actions/runs/373225621

Using pin=pin_speaker is no longer affecting music.pitch(440) in the sample above.

@dpgeorge After a reset music.play(music.BLUES, pin=pin_speaker) just plays on speaker, but after music.pitch it plays on both.

```
    music.pitch(440)
    music.stop()
    music.play(music.BLUES, pin=pin_speaker)

```

dpgeorge, 2020-11-21:
> After a reset music.play(music.BLUES, pin=pin_speaker) just plays on speaker, but after music.pitch it plays on both.

Thanks for the report.  Should be fixed by 2f70b757f3bd39454d16b6d077aa078f6b100397

martinwork, 2020-11-23:
Thanks @dpgeorge . That's fixed it.

microbit-carlos, 2020-11-23:
Thanks Martin!

closed 2020-11-23 by microbit-carlos