microbit-foundation/micropython-microbit-v2 #25: Music and LCD problem

crouse12, 2020-11-14:
Please watch the following code.
[image]
I found that music will play after LED screen show 'Hello, World!'.
I want they play at the same time.
Is it possible?

microbit-carlos, 2020-11-16:
Hi @crouse12,
You can use the `wait` argument in scroll to do that: `display.scroll('Hello, World!', wait=False)`
https://microbit-micropython.readthedocs.io/en/latest/display.html

crouse12, 2020-11-16:
Thanks for your help. I will test it.
Can music.play execute in the background, too?
I try to use micropython to create scratch 3.0 blocks for microbit v2 now.
music.play and display.scroll usually make some mistakes for my scrath 3.0 blocks, when i read movement and gestures at the same time.
https://www.youtube.com/watch?v=hKWrgvlkIow&t=6s

microbit-carlos, 2020-11-16:
https://microbit-micropython.readthedocs.io/en/latest/music.html#music.play

Although there is some undergoing work right now with all the sound-generating stuff and some things might not work yet.

crouse12, 2020-11-16:
Thanks.
I will wait them.

microbit-carlos, 2020-11-23:
This should be working in the latest https://python.microbit.org/v/beta (beta.7).
If you find any issues could you let us know by opening new bug reports? Thanks!

closed 2020-11-23 by microbit-carlos

crouse12, 2020-11-23:
> This should be working in the latest https://python.microbit.org/v/beta (beta.7).
> If you find any issues could you let us know by opening new bug reports? Thanks!

The latest python editor solves this problem. Thanks.