microbit-foundation/micropython-microbit-v2 #26: sonar sensor problem(hcsr04)

crouse12, 2020-11-22:
The following code for hcsr04 can work on microibit v1, but microbit v2 does not.
Python editor version for my test is 2.1.0-beta.7 for hcsr04.
I think time_pulse_us or sleep_us has some bugs.

from microbit import *
from utime import sleep_us
from machine import time_pulse_us
def sonar():
    pin2.write_digital(0)
    sleep_us(10)
    pin2.write_digital(1)
    sleep_us(10)
    pin2.write_digital(0)
    pin1.set_pull(pin1.PULL_DOWN)
    t = time_pulse_us(pin1, 1, 10**5)
    return t // 58

while True:
    display.scroll(str(sonar()))
    sleep(1000)

microbit-carlos, 2020-11-23:
What's the result you get in a v1 vs v2?

crouse12, 2020-11-23:
microbit v1 show distance on led screen.
However, microbit v2 does not show anything.
I found that makecode has the same issue.

https://github.com/microsoft/pxt-microbit/issues/3731

crouse12, 2020-12-07:
I find v0.2.23 of codal-microbit-v2 now released.
It solves pulse problem.
Does micropython update, too?

https://github.com/microsoft/pxt-microbit/issues/3731

dpgeorge, 2020-12-14:
Yes MicroPython is now updated to codal v0.2.23.

jaustin, 2020-12-14:
@crouse12 could you please test with the build from https://github.com/microbit-foundation/micropython-microbit-v2/actions/runs/419939816 (this is an automatically built version from a commit on master, so isn't exactly what the next release will look like, but does have the updated CODAL).

crouse12, 2020-12-14:
I test time_pulse_us on microbit v2 using the following two MicroPython Versions. 
However, they cannot work.
I think there are something wrong on time_pulse_us.

[image]

[image]

jaustin, 2021-01-20:
@dpgeorge does the patch from @xmeow look okay to you? Would be great to close this off with the next release if so?

microbit-carlos, 2021-02-05:
Looks like PR https://github.com/microbit-foundation/micropython-microbit-v2/pull/58 is adding the CODAL method to the MicroPython pin API. I think ultimately what we need is to make sure `machine.time_pulse_us` works on V1.
MakeCode also had an issue in this area, and the feature was also implemented in the MakeCode code, so the functionality has been moved to CODAL. It'd be good to use CODAL in this case as well to unify the implementation.

dpgeorge, 2021-02-08:
`machine.time_pulse_us()` has been fixed by b7a76e449cf1b437972c21efaf468fbe5891c6a2 .  So this function should now work the same as on V1.

> It'd be good to use CODAL in this case as well to unify the implementation.

If the built-in MicroPython one works, is it ok to just keep that the way it is?

jaustin, 2021-02-12:
I think if it's not a trivial switch over, I'm happy to leave things as they are with the V2 implementation.

microbit-carlos, 2021-02-17:
Great, closing this as fixed. 🎉

closed 2021-02-17 by microbit-carlos