microbit-foundation/micropython-microbit-v2 #28: music.play stops music.pitch, but music.pitch doesn't stop music.play

microbit-carlos, 2020-11-25:
Reported by @martinwork in https://github.com/microbit-foundation/micropython-microbit-v2/issues/15#issuecomment-732080446:

> ```python
> music.play(music.BLUES, wait=False)
> music.pitch(440)
> music.stop()
> ```
> I noticed that music.play cancels any previous music.play or music.pitch, but music.pitch doesn't cancel a previous music.play. The music.pitch sounds briefly, and the blues keep playing. Is this expected?

dpgeorge, 2020-12-07:
Should be fixed by a76e1413bcd66f128a31d98756fc3d1f336d1580

closed 2020-12-07 by dpgeorge

martinwork, 2020-12-07:
Thanks @dpgeorge  This works as expected with the hex from https://github.com/microbit-foundation/micropython-microbit-v2/actions/runs/404776108

```
from microbit import *
import music

while True:
    music.pitch(440)
    sleep(1000)
    music.play(music.BLUES, wait=False)
    sleep(1000)

```