microbit-foundation/micropython-microbit-v2 #29: Changing the baudrate to 115200 (default rate with interface chip) captures odd characters

microbit-carlos, 2020-11-25:
The default serial connection between MicroPython and DAPLink should be 115200, setting this baud rate in the REPL via `uart.ini()`causes odd characters to be captured/sent-back. GIF below.

[image]

microbit-carlos, 2020-12-18:
CODAL has been updated with some serial fixes, so I've updated the CODAL version here in a branch, and while some of the other problems are fixed, this one is still not resolved.

https://github.com/microbit-foundation/micropython-microbit-v2/actions/runs/430344171

dpgeorge, 2020-12-21:
> I've updated the CODAL version here in a branch

Thanks, I've now merged that (it fixes the "serial garbage on start" up issue).

>  this one is still not resolved.

Yes, confirmed this issue here is not fixed.  I did manage to reproduce it using pure C++ by adding a call to redirect (which is what MicroPython does):
```c
#include "MicroBit.h"

MicroBit uBit;

int  main() {
    uBit.init();

    for (int i = 0; i < 10; ++i) {
        char x = uBit.serial.read();
        uBit.serial.sendChar(x);
    }

    uBit.serial.setBaud(115200);
    uBit.serial.redirect(uBit.io.usbTx, uBit.io.usbRx);

    while (1) {
        char x = uBit.serial.read();
        uBit.serial.sendChar(x);
    }
}
```
In the above test, the first 10 chars are echoed back correctly, but then when it redirects (to the same pins) following chars are not.

microbit-carlos, 2020-12-21:
Thanks Damien, I've open https://github.com/lancaster-university/codal-microbit-v2/issues/60 in the CODAL repo.

dpgeorge, 2021-02-08:
Looks like there has been no progress yet in the CODAL on this issue.

microbit-carlos, 2021-03-08:
Issue has been fixed as part of https://github.com/lancaster-university/codal-microbit-v2/issues/60 and will available in the next CODAL tag (coming soon).

dpgeorge, 2021-03-16:
CODAL was updated to v0.2.25 in 76054c2e9445d8eea285132c2374a18ddc690514 so hopefully this issue is fixed now.

microbit-carlos, 2021-03-16:
Can confirm this is fixed and will be present in the beta.5 release, thanks Damien!

closed 2021-03-16 by microbit-carlos