microbit-foundation/micropython-microbit-v2 #31: music.stop pin argument

microbit-carlos, 2020-11-26:
The `music.stop()` function can take a pin argument like `play()`.

Does it currently correctly process tuples? It doesn't throw an error, but don't know if it works as expected.
Technically in this case the user could also do `music.stop(pin=pin_speaker)` and `music.stop(pin=pin0)` to stop the music in both pins, but it does make sense to mirror play.

I guess my question is how having music play over multiple pins is going to interact with the stop function. Will it work when doing `play(..., pin=(pin_speaker, pin0))` and then we only want to stop a single pin `music.stop(pin=pin0)`? 

This on the REPL currently stops the music playing in the speaker
```
>>> music.play(music.PYTHON, wait=False)
>>> music.stop(pin0)
```

Another thing to consider for https://github.com/microbit-foundation/micropython-microbit-v2/issues/30 as well.

jaustin, 2020-12-14:
New behaviour
* no argument: stop all pins that are playing (do nothing if no music playing)
* pin argument: stop on speaker and that pin if playing, do not stop on other pins.

This makes the pin argument mostly redundant, but kept for compatibility and the subtle case that you're, say, playing on pin2, and call stop on pin1 (speaker will stop in any case).

microbit-carlos, 2020-12-21:
Related: https://github.com/microbit-foundation/micropython-microbit-v2/issues/50

jaustin, 2021-01-20:
@dpgeorge did the stop behaviour described here get implemented in the end?

dpgeorge, 2021-02-08:
> did the stop behaviour described here get implemented in the end?

No.  The current behaviour is simply that, whether or not `music.stop()` is given a pin argument, it stops the music and disconnects whatever pin was used from the mixer (hence also stopping all other sounds like expressions on that pin).  The speaker stays connected though.

You can test this behaviour with:
```
music.play(music.NYAN,wait=0); audio.play(Sound.HAPPY); sleep(50); music.stop()
```
(that will play music for 50ms then stop it, but the HAPPY will continue, on the speaker)

dpgeorge, 2021-02-08:
> pin argument: stop on speaker and that pin if playing, do not stop on other pins

This is not really possible to do with the current CODAL, because you can only play on one pin at a time (in addition to the speaker).

dpgeorge, 2021-02-08:
This issue is affected by the outcome of #50, so maybe that needs to be resolved first?

dpgeorge, 2021-06-29:
This was implicitly fixed as part of #50 : `music.stop()` now only stops any ongoing sequencing of notes, it does not disconnect any pins.

The following test should now work as expected, the music stops but the sound expression continues (on both the internal speaker and external pin0):
```python
music.play(music.NYAN,wait=0); audio.play(Sound.HAPPY,wait=0); sleep(50); music.stop()
```

dpgeorge, 2021-06-29:
For completeness, this is the commit that addressed the issue here: 6790bb511b7966e727c147ff27269121e340435f

closed 2021-06-29 by dpgeorge