microbit-foundation/micropython-microbit-v2 #32: Problems building in Ubuntu 20.04

aivarannamaa, 2020-11-27:
I followed the instructions in README.md. When doing `make` in `src`, I got following errors:

```
In file included from ../../lib/micropython/py/qstr.h:30,
                 from ../../lib/micropython/py/lexer.h:32,
                 from ../../lib/micropython/py/compile.h:29,
                 from ../../lib/micropython/py/builtinimport.c:32:
../../lib/micropython/py/builtinimport.c: In function 'mp_builtin___import__':
./mpconfigport.h:35:50: error: 'PATH_MAX' undeclared (first use in this function)
   35 | #define MICROPY_ALLOC_PATH_MAX                  (PATH_MAX)
      |                                                  ^~~~~~~~
../../lib/micropython/py/misc.h:169:63: note: in definition of macro 'VSTR_FIXED'
  169 | define VSTR_FIXED(vstr, alloc) vstr_t vstr; char vstr##_buf[(alloc)]; vstr_init_fixed_buf(&vstr, (alloc), vstr##_buf);
      |                                                              ^~~~~

../../lib/micropython/py/builtinimport.c:360:22: note: in expansion of macro 'MICROPY_ALLOC_PATH_MAX'
  360 |     VSTR_FIXED(path, MICROPY_ALLOC_PATH_MAX)
      |                      ^~~~~~~~~~~~~~~~~~~~~~
./mpconfigport.h:35:50: note: each undeclared identifier is reported only once for each function it appears in
   35 | #define MICROPY_ALLOC_PATH_MAX                  (PATH_MAX)
      |                                                  ^~~~~~~~
../../lib/micropython/py/misc.h:169:63: note: in definition of macro 'VSTR_FIXED'
  169 | define VSTR_FIXED(vstr, alloc) vstr_t vstr; char vstr##_buf[(alloc)]; vstr_init_fixed_buf(&vstr, (alloc), vstr##_buf);
      |                                                              ^~~~~

../../lib/micropython/py/builtinimport.c:360:22: note: in expansion of macro 'MICROPY_ALLOC_PATH_MAX'
  360 |     VSTR_FIXED(path, MICROPY_ALLOC_PATH_MAX)
      |                      ^~~~~~~~~~~~~~~~~~~~~~
../../lib/micropython/py/builtinimport.c:360:16: error: unused variable 'path_buf' [-Werror=unused-variable]
  360 |     VSTR_FIXED(path, MICROPY_ALLOC_PATH_MAX)
      |                ^~~~
../../lib/micropython/py/misc.h:169:51: note: in definition of macro 'VSTR_FIXED'
  169 | #define VSTR_FIXED(vstr, alloc) vstr_t vstr; char vstr##_buf[(alloc)]; vstr_init_fixed_buf(&vstr, (alloc), vstr##_buf);
      |                                                   ^~~~
cc1: all warnings being treated as errors
make[1]: *** [../../lib/micropython/py/mkrules.mk:63: build/py/builtinimport.o] Error 1
make[1]: Leaving directory '/home/annamaa/python_stuff/mibi2/src/codal_port'
make: *** [Makefile:44: libmicropython] Error 2
```

microbit-carlos, 2020-11-30:
Hi Aivar,
What OS and version of GCC are you using? Any other system information would be quite useful as well.

microbit-carlos, 2020-11-30:
@aivarannamaa this looks like the same issue as https://github.com/microbit-foundation/micropython-microbit-v2/issues/5#event-4051458787, so let's merge the conversation. Could you reply to my last message over there?

closed 2020-11-30 by microbit-carlos