microbit-foundation/micropython-microbit-v2 #33: microphone.is_event() stops responding after a few events

martinwork, 2020-12-01:
This program sees LOUD at startup. After 3 more loud events, it's stuck on QUIET.

```
from microbit import *

display.clear()

while True:
    if microphone.is_event(SoundEvent.LOUD):
        display.show(Image.SQUARE)
    elif microphone.is_event(SoundEvent.QUIET):
        display.show(Image.SQUARE_SMALL)
    else:
        display.clear()

```

microbit-carlos, 2020-12-01:
Thanks Martin, I can replicate this in MicroPython 2.0.0-beta.1.

Could you try the same programme in CODAL C++ and see if it replicates there as well?

dpgeorge, 2020-12-02:
This is related to the semantics of microphone events, #36 .  The point is that the event history (L from that issue) gets full and so no more events come in and it gets stuck (until microphone.get_events() is called).

To make this work try:
```python
from microbit import *

display.clear()

while True:
    if microphone.was_event(SoundEvent.LOUD):
        display.show(Image.SQUARE)
    elif microphone.was_event(SoundEvent.QUIET):
        display.show(Image.SQUARE_SMALL)
```

martinwork, 2020-12-02:
I don't know how to replicate this in CODAL. I've only seen the LEVEL_THRESHOLD_HIGH/LOW events and can test "is loud" with levelDetector->getValue() > levelDetector->getHighThreshold().

I was testing is_event, so can't use was_event instead!

I was expecting is_event and was _event to behave like button is_pressed and was_pressed, but looking at #36, I think the test shows is_event is behaving as intended.

closed 2020-12-02 by martinwork

dpgeorge, 2020-12-14:
This should be fixed by c57910ce1d6957dda43d5a8e4596d511be20b446

martinwork, 2020-12-14:
Thanks @dpgeorge . It doesn't get stuck now.