microbit-foundation/micropython-microbit-v2 #35: uart problem

crouse12, 2020-12-01:
I found that uart has a problem if I follow the following steps.
(1)
Connect to microbit v2 by using putty,
(2)
Reset the microbit v2's power.
(3)
Connect to microbit v2 by using putty, 
You can find that microbit v2 cannot work correctly.

https://youtu.be/ZrAsxoCKZTo

dpgeorge, 2020-12-02:
Yes I have seen that issue before, that sometimes on initial connection the REPL is not responsive and just returns garbage characters.

@microbit-carlos I'm not sure where the issue lies, maybe with DAP link?

The fix is to press the reset button on the back of the microbit.

crouse12, 2020-12-02:
Can you clear buffers or session when the Microbit v2 start?

crouse12, 2020-12-07:
Can anyone help me to solve uart problem?

martinwork, 2020-12-07:
Problem exists with a C++ test too.

dpgeorge, 2020-12-07:
> Problem exists with a C++ test too.

Can you please post C++ code that reproduces it?  I guess it's an issue with the CODAL and initialisation of the serial UART?

martinwork, 2020-12-08:
@dpgeorge https://github.com/lancaster-university/codal-microbit-v2/issues/45

dpgeorge, 2020-12-14:
See also https://github.com/lancaster-university/codal-microbit-v2/issues/55

dpgeorge, 2020-12-21:
Fixed by df3caf7176188303686801a1f1fe6e6496d346ad

closed 2020-12-21 by dpgeorge