microbit-foundation/micropython-microbit-v2 #36: microphone.is_event() question

microbit-carlos, 2020-12-01:
Just to make sure this is correct:

- `was_event(event)` returns true if the event was captured (in any order) since the last time `was_event()` or `get_events()` was called, and clears the history
- `is_event(event)` only returns true if the last captured event in the list matches, no matter how long ago it was, and it doesn't clear the history
- `current_event()` returns the event that is happening "right now", which is also added to the event list and doesn't clear it.

dpgeorge, 2020-12-02:
There are two kinds of histories:
- S - whether any given event was capture (a set)
- L - the order of captured events (a list).

`was_event(event)` will return true if that event is in S, and remove that event from S.  It will also clear L.

`is_event(event)` returns true if that event is the most recent (last) event in L.

`get_events()` returns a copy of L and clears L.

 Not sure if that's all sensible, but that's how v1 worked :)

martinwork, 2020-12-02:
I was thinking was_event and is_event would behave like button was_pressed and is_pressed. 

How does current_event relate to S and L?

Would it be helpful to have was_loud, was_quiet, is_loud and is_quiet, that do behave like the button functions?

martinwork, 2020-12-02:
I don't know what L is used for. Could it be a circular buffer that records the most recent events, and doesn't get stuck?

dpgeorge, 2020-12-14:
The semantics of `microphone.is_event()` have been improved in c57910ce1d6957dda43d5a8e4596d511be20b446 so it doesn't get blocked by the history filling up.

Test code:
```python
while True:
    print(microphone.is_event(SoundEvent.LOUD), microphone.is_event(SoundEvent.QUIET))
```
That should swap between `False True` and `True False` when noise is made.

jaustin, 2020-12-14:
Thanks Damien, that sounds perfect.

closed 2020-12-14 by jaustin