microbit-foundation/micropython-microbit-v2 #37: Speech quality

dpgeorge, 2020-12-02:
The speech quality has recently been improved in 081bb4e7ecc418a768292367a8f8db3ffaf8d082 which adds double buffering to the output.  Before this change the speech was very "rough" sounding, I think because there were gaps between the chunks going out to the audio mixer.  With double buffering the speech output is much better and sounds the same as playing a whole speech sample as one big buffer.

Please test and compare with a micro:bit v1!

mytechnotalent, 2020-12-02:
I will test @dpgeorge as I demo my Study Buddy featuring the bot called "Mr. George" named after you of course at the big micro:bit LIVE global conference on Friday. - https://github.com/mytechnotalent/MicroPython-micro-bit_Study_Buddy here.

I will test this today.  Do I need to build from within the repo or is this pushed up to the MP Web Editor here - https://python.microbit.org/v/beta

The issue I found was the speech was peaking or overdriving slightly.

mytechnotalent, 2020-12-02:
Demo of change 1 - up close 
```
from microbit import display, Image
from speech import say
display.show(Image.SURPRISED)
say('Hello, my name is Mister George.')
say('It is so great to meet you all!')
say('I am ex si ted to help you change the world!')
display.show(Image.HAPPY)
```
[FIXED_081bb4e.MOV.zip](https://github.com/microbit-foundation/micropython-microbit-v2/files/5630308/FIXED_081bb4e.MOV.zip)

Demo of change 1 - farther back
```
 from microbit import display, Image
from speech import say
display.show(Image.SURPRISED)
say('Hello, my name is Mister George.')
say('It is so great to meet you all!')
say('I am ex si ted to help you change the world!')
display.show(Image.HAPPY)
```

mytechnotalent, 2020-12-02:
[Uploading FIXED_081bb4e-BACK.MOV.zip…](Farther back demo.)

microbit-carlos, 2020-12-03:
Thanks Damien, I gave this a quick spin and sounds much better!

I've tagged and released beta.2 to deploy to the https://python.microbit.org/v/beta (still deploying) so it'll be easier for other people to test as well.

closed 2020-12-14 by jaustin