microbit-foundation/micropython-microbit-v2 #39: Backspace correction echo fails in Mu Repl

AllegroGavin, 2020-12-08:
I'm using beta.2 with Mu 1.1.0.alpha.2 (have not tried beta.1) and frequently when trying to use backspace to correct input in the repl it will echo specific garbage. For example, here I have typed 12345678 and then hit backspace ~4 times, the first three times it works correctly, then it spits out the (crossed box)[K sequence. The repl received and acted on the backspace.

[image]

It's always this same three character sequence, and it tends to occur in bursts, i.e. I can hold down backspace and maybe for 15 characters it's fine but then I get a burst of the three characters a few times in a row. It only appears to be the backspace that shows a visible issue.

I presume it's trying to output (backspace)(space)(backspace) but is glitching somehow in the serial data stream etc.

martinwork, 2020-12-09:
I can hit problems with backspace in the Mu REPL quite easily, but not in the python.microbit.org REPL or tera term.

dpgeorge, 2020-12-14:
This seems like a problem with Mu's handling of escape sequences.

jaustin, 2020-12-14:
@microbit-carlos is going to see if this is also visible on V1 with Mu - anyone who has seen it there please shout!

microbit-carlos, 2020-12-14:
I gave this a go with the Mu alpha.2 release (same as @AllegroGavin, which doesn't include some of the latest changes in how it processes space sequences) and I can easily reproduce this in V2 but not in V1.

In this crude test I just type `asd` as fast as i can while pressing the backspace key as well.
In V2 we can see at least 7 cases where the scape sequence is missing something and printed back:

[image]

V1 with twice the "samples" i couldn't get a single hit:

[image]

microbit-carlos, 2020-12-14:
Actually an easier way to replicate is to fill the REPL line with text and then keep the backspace key pressed.

The incomplete scape sequence gets sometimes printed and immediately deleted, but it looks like it gets correctly processed inside the MicroPython buffer, since at some point it stops removing characters and the history (up arrow) shows an empty buffer:

[image]

Could not replicate this with V1.

dpgeorge, 2020-12-15:
This could be related to #15

microbit-carlos, 2020-12-15:
In the end this was a problem on Mu that was somehow easier to trigger with V2.

The escape characters processing has been updated in the Mu repository, but there isn't an official release yet.
From the work that was carried out at the time, I think this is to do with the fact that Mu used to break up the serial processing in blocks of 20 bytes and that could divide the bytes of a scape sequence.

@AllegroGavin if you like you test a Windows CI build (down in the artifacts section) and that should should not present the issue anymore: https://github.com/mu-editor/mu/actions/runs/419131068

closed 2020-12-15 by microbit-carlos

AllegroGavin, 2020-12-15:
Yes, I can confirm the problem is no longer detectable using the new Mu build artifact version.

Thanks for looking into it!