microbit-foundation/micropython-microbit-v2 #40: uart.any() seems to be always true on V2

martinwork, 2020-12-08:
On V2 the program below immediately shows A on the display, then E, indicating `uart.any()` has returned true and the program is waiting inside uart.read(). Typing a single character in tera term shows F, G and returns straight to E, waiting in uart.read() again.

On V1 it just flashes the pixel, indicating that `uart.any()` is returning false, until a character is typed in tera term.

In the Python editor, https://python.microbit.org/v/beta, the word "any" is displayed green. Is that expected?

On V2 at least, uart.read() seems never to timeout.

In a C++ CODAL program, uBit.serial.isReadable() seems to work, so long as setRxBufferSize() is called:
https://github.com/lancaster-university/codal-microbit-v2/issues/47

This program also demonstrates the V2 power up garbage problem. After connecting with tera term, power cycle V2 a few times, until it shows E,F,G on startup, then type SPACE in tera term to see odd characters echoed back.

```
from microbit import *
import time

def echo():
    if uart.any():
        display.show("E")
        sleep(200)
        bytes_in = uart.read(1)
        display.show("F")
        sleep(200)
        if len(bytes_in) > 0:
            uart.write( bytes_in)
        else:
            uart.write(b'?')
        display.show("G")
        sleep(200)    
        display.clear()
            
def toggle_pixel( last_time):
    if time.ticks_ms() - last_time > 1000:
        display.set_pixel(0,0,0 if display.get_pixel(0,0) else 9)
        last_time = time.ticks_ms()
    return last_time
        
#uart.init(baudrate=115200, bits=8, parity=None, stop=1, *, tx=None, rx=None)
uart.init(baudrate=115200)

if uart.any():
    display.show("A")
    sleep(100)
    display.clear()
    
toggle_time = time.ticks_ms()

while True:
    toggle_time = toggle_pixel(toggle_time)  
    echo()
    sleep(0)

```

dpgeorge, 2020-12-14:
Thanks for the report, I could confirm the issue with `uart.any()` always returning True, as well as `uart.read(1)` never timing out.

Both these issues should be fixed by 967c0765c88f3b1fb3e765123bf0ae788bac1bb3

The garbage characters is a separate (CODAL?) issue.

jaustin, 2020-12-14:
@martinwork can you please test with the fix and close this issue to confirm it is fixed?

martinwork, 2020-12-14:
Testing with 967c076 and Mu, uart.any and the sample above work as expected.

closed 2020-12-14 by martinwork