microbit-foundation/micropython-microbit-v2 #41: Expose @micropython.asm_thumb to allow in-line ASM for V2

mytechnotalent, 2020-12-10:
It is extremely important that we have access to inline assembly to add customizations for example:

```
from microbit import *

display.show('S')

def get_serial_number(type=hex):
    NRF_FICR_BASE = 0x10000000
    DEVICEID_INDEX = 25 # deviceid[1]

    @micropython.asm_thumb
    def reg_read(r0):
        ldr(r0, [r0, 0])
    return type(reg_read(NRF_FICR_BASE + (DEVICEID_INDEX*4)))
    
while True:
    if button_a.was_pressed():
        display.scroll(get_serial_number())
        sleep(1000)
        display.show('S')
        
    sleep(100)
```

This allows inline assembly to be able to be integrated into the V2.  Can we expose the flag such that we will be able to run the above code and the like @microbit-carlos?

dpgeorge, 2020-12-11:
micro:bit v1 had support for inline assembly, so we should add it to v2 (and it is very useful and a good teaching tool!).

But for the example above there's a simpler way to do it:
```python
import machine
machine.mem32[NRF_FICR_BASE + DEVICEID_INDEX * 4]
```

mytechnotalent, 2020-12-11:
Thanks @dpgeorge it really will be a good teaching tool as I teach ARM32 Reverse Engineering and I would love to do some work with the v2 in this regard as well.

You are correct working with the machine module is simpler however I am working with the bitio module which interfaces with CPython for AI/ML applications for the v2.  This architecture does not have the machine module available to it so having the inline assembly allows myself and others to integrate any functionality at all with ease.

dpgeorge, 2020-12-14:
Enabled in ca485d2ccf79801d1d0761a9d126ddf7cf680d4d

closed 2020-12-14 by dpgeorge