microbit-foundation/micropython-microbit-v2 #42: machine.Timer

godsic, 2020-12-12:
Hi there,

Any plans to implement `machine.Timer` class that might be beneficial for, e.g., uniform data sampling, DSP, etc.?

jaustin, 2021-01-20:
@godsic we haven't currently got plans to do this, but we're planning to do some work around data sampling, etc over the next few months. 

It would really help if you have the time to write a few examples of where and how you think this would help (see for example #49 where we are using some API examples to work out what the right thing to implement is.)

yanghao, 2021-04-20:
This is particular useful for non-blocking bit-sampling or big-banging, for example, if you want to implement an IR receiver using bit-sampling in a non-blocking way (event-driven way), it is currently hard to do so ... also any plans to upstream the v2 support like v1 directly to micropython? 

https://github.com/1010Technologies/pxt-makerbit-ir-receiver/blob/060abe9672b5c9f0eeb281b75997651fa69d8f9a/infrared-receiver.ts#L135