microbit-foundation/micropython-microbit-v2 #43: Errors while compiling on Ubuntu 20.04

FalconDQ, 2020-12-13:
When compiling on UBUNTU I get these errors:
[  0%] Building CXX object libraries/codal-core/CMakeFiles/codal-core.dir/source/core/CodalCompat.cpp.o
c++: warning: ‘-mcpu=’ is deprecated; use ‘-mtune=’ or ‘-march=’ instead
c++: error: unrecognized command line option ‘-mthumb’
c++: error: unrecognized command line option ‘-mfpu=fpv4-sp-d16’
c++: error: unrecognized command line option ‘-mfloat-abi=softfp’

microbit-carlos, 2020-12-16:
Hi @FalconDQ,
Have you got installed `gcc-arm-none-eabi`?

FalconDQ, 2020-12-17:
thank you or the reply! I checked, it is: gcc-arm-none-eabi is already the newest version (15:9-2019-q4-0ubuntu1).

microbit-carlos, 2020-12-18:
Thanks @FalconDQ. It still sounds like it might not be using the arm version of gcc.
It would be quite useful if you could start from 0 and copy/paste here every single command and output (all of it, including the commands).

FalconDQ, 2020-12-19:
Thanks for your support! I fixed it because of your help! I'm not that great with Linux and C and make and compiling stuff, let alone cross-compiling. I installed Linux in a virtual machine just for the purpose of compiling Micropython. I think somewhere along the way I installed Cmake because of an error I got. Later on I installed gcc-arm-none-eabi on top of that. All based on Google "research". Only after removing both Cmake and gcc-arm-none-eabi and   
clone this Git repository and redoing all the steps it compiled without errors.  Maybe it's worth mentioning in the README that installing gcc-arm-none-eabi compiler is required to compile this. Just to make it more NOOB proof! ;-). Thanks again!

closed 2020-12-19 by FalconDQ