microbit-foundation/micropython-microbit-v2 #44: machine.time_pulse_us() makes V2 unresponsive

microbit-mark, 2020-12-14:
> micro:bit support: #41888

From user:

This code works as expected on V1, but makes the V2 unresponsive

``` Python
import machine
dir(machine)
['__name__', 'unique_id', 'reset', 'freq', 'disable_irq', 'enable_irq', 'mem8', 'mem16', 'mem32', 'time_pulse_us']
machine.time_pulse_us(pin0,1)
-2
```

The -2 timeout is expected with nothing connected to the pin

using MicroPython v1.13 on 2020-11-24; micro:bit v2.0.0-beta.1 with nRF52833.

This code is used in French Python curriculum materials when the micro:bit is connected to an ultrasonic sensor.

Maybe related to https://github.com/microbit-foundation/micropython-microbit-v2/issues/26

microbit-carlos, 2020-12-14:
So does the output shown in your code snippet belong to a V1?
Apologies, when you mentioned this before I thought V2 was returning `-2` which is why I though it was different that issue #26 (where V2 hangs).

If that is the case we can merge this into #26. If it's not clear if that's the case could you confirm this with the user?

crouse12, 2020-12-15:
This issue is the same as
https://github.com/microbit-foundation/micropython-microbit-v2/issues/26

microbit-mark, 2020-12-15:
I think the snippet may belong to V1. The timeout does not reproduce for me on V2, but the device becoming unresponsive does. Closing in favour of #26

closed 2020-12-15 by microbit-mark