microbit-foundation/micropython-microbit-v2 #52: Continuously sending bytes crash

onionstack, 2020-12-26:
Send 100 bytes at a time, and it will crash if you send dozens of times continuously

```
[18:12:21.153]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.167]IN←◆456789012345678901234567890123456789012345678901234567890123456789
>>> 0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789
12345678901234567890123456789012345678901234567890123456789
[18:12:21.177]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□0123456789012345678901234567890123456789
>>> 0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789
123456789012345678901
[18:12:21.199]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.210]IN←◆234567890123456789012345678901234567890123456789012345678901234567890123456789
>>> 0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789
123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789
>>> 01234567890123456789012345678901234567890123456
[18:12:21.232]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.247]IN←◆78901234567890123456789012345678901234567890123456789
123456789012345678901234567890123456789012345678901234567890123456789012345678
[18:12:21.262]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.272]IN←◆901234567890123456789
>>> 0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789
1234567890
[18:12:21.284]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.295]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.309]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.327]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.342]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.358]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.373]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.389]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.405]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.421]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.438]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.452]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.467]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
[18:12:21.483]OUT→◇0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789□
```

carlosperate, 2020-12-26:
What version of MicroPython are you using? Is this problem still present in the beta.3? https://github.com/microbit-foundation/micropython-microbit-v2/releases/tag/v2.0.0-beta.3

onionstack, 2020-12-29:
@carlosperate I use the beta.3 version and the problem still exists. [MicroPython v1.13 on 2020-12-21; micro:bit v2.0.0-beta.3 with nRF52833](https://github.com/microbit-foundation/micropython-microbit-v2/releases/tag/v2.0.0-beta.3)

jaustin, 2021-01-20:
As far as continuing to triage this, we need to see whether we can reproduce this in C++ (IE in CODAL). @microbit-carlos let's see if we can reproduce this.

microbit-mark, 2021-02-04:
@martinwork Can you try and recreate this as per Jonny's comment https://github.com/microbit-foundation/micropython-microbit-v2/issues/52#issuecomment-763828695

martinwork, 2021-02-10:
@onionstack 
I have created issue https://github.com/lancaster-university/codal-microbit-v2/issues/80
Does this seem likely to be the same problem? Or have I misunderstood?

jaustin, 2021-02-12:
@onionstack could you please clarify what you mean by 'crash'? Do you mean lock up? Or throw an OS error? Or corrupted data? Would be good to knoe for sure. If you have a Python file that reproduces this please share with us so we can test with an updated version of MicroPython.

microbit-mark, 2021-02-19:
A similar problem is reported in micro:bit support ticket 43114 with this reproducer:

The code below will run indefinitely on V1. However, on V2, it will run for ~300 iterations and then lock-up with the display going off. After pressing the reset, the program will continue running for another brief period and so on.

``` Python
from microbit import *
 
while True:
  if button_a.is_pressed():
    display.show(Image.MEH )
  elif button_b.is_pressed():
    display.show(Image.SAD )
  else:
    display.show(Image.HAPPY )
 ```

microbit-mark, 2021-03-04:
@martinwork Are you able to try and replicate this in MicroPython so we can rule in/out lancaster-university/codal-microbit-v2#80

microbit-carlos, 2021-03-04:
Yeah if lancaster-university/codal-microbit-v2#80 is worse in V1 than V2 we should try to replicate this in MicroPython. My guess is that this might be replicable in V2 and not V1? In that case we are likely looking at different issues.

martinwork, 2021-03-04:
Do we want a program to send bytes to the REPL and check the echo?

microbit-carlos, 2021-03-04:
Yes, that sounds like a good test. To be fair it sounds like V2 crashes anyway, so if we first give it a quick try only from the micro:bit side that might be good enough to get an initial replication.

martinwork, 2021-03-04:
Using tera term to paste/Enter/paste/Enter to the REPL (hex flashed from beta) crashes V2, but not always easily.

After Ctrl-E, I couldn't make it crash. I couldn't make it crash with V1.

`1234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890
`

With the program below, I tested V1 and V2 REPL, with just a Ctrl-C and with Ctrl-E too.

With Ctrl-C on it's own V1 survives, but V2 crashes

After Ctrl-E, V1 crashes quickly. V2 survives a bit longer than with just Ctrl-C, failing with the message "FATAL: uncaught NLR 20004ff0".

Adding a sleep between each line delays the crashes, but doesn't seem to stop them.

```
#!/usr/bin/env python

#
# The MIT License (MIT)
# 
# Copyright (c) 2020 Micro:bit Educational Foundation.
# 
# Permission is hereby granted, free of charge, to any person obtaining a
# copy of this software and associated documentation files (the "Software"),
# to deal in the Software without restriction, including without limitation
# the rights to use, copy, modify, merge, publish, distribute, sublicense,
# and/or sell copies of the Software, and to permit persons to whom the
# Software is furnished to do so, subject to the following conditions:
# 
# The above copyright notice and this permission notice shall be included in
# all copies or substantial portions of the Software.
# 
# THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
# IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
# FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL
# THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
# LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
# FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER
# DEALINGS IN THE SOFTWARE.
#

# This is a client for the micropython REPL

import subprocess
import shutil
import os
import stat
import distutils.dir_util
import filecmp
import json
import sys
import time

import serial
import serial.tools
import serial.tools.list_ports

LOOPS = 10000

PORT=''
#PORT='COM5'

BAUDRATE = 115200
#BAUDRATE = 57600
#BAUDRATE = 9600

linebreakOut = b'\r'

ctrlC = b'\x03'
ctrlE = b'\x05'

bytesOut = bytearray()
for i in range(32,255):
    bytesOut.append(i)
bytesOut.append(10)

bytesOut10 = b'1234567890'
bytesOut26 = b'ABCDEFGHIJKLMNOPQRSTUVWXYZ'
bytesOut30 = b'ABCDEFGHIJKLMNOPQRSTUVWXYZ1234'
bytesOut31 = b'ABCDEFGHIJKLMNOPQRSTUVWXYZ12345'
bytesOut32 = b'ABCDEFGHIJKLMNOPQRSTUVWXYZ123456'
bytesOut62 = b'1234567890ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz'
bytesOut100 = b'123456789a123456789b123456789c123456789d123456789e123456789f123456789g123456789h123456789i123456789j'
bytesOut250 = b'123456789a123456789b123456789c123456789d123456789e123456789f123456789g123456789h123456789i123456789j123456789k123456789l123456789m123456789n123456789o123456789p123456789q123456789r123456789s123456789t123456789u123456789v123456789w123456789x123456789y'
bytesOut = bytesOut100;

port = PORT

if len(port) == 0:
  listPortInfo = serial.tools.list_ports.comports(include_links=False)
  port = listPortInfo[0].device

print("PORT " + port)

#############################
ser = serial.Serial()
ser.port = port
ser.baudrate = BAUDRATE
ser.bytesize = 8
ser.parity = 'N'
ser.stopbits = 1
ser.timeout = 1
ser.xonxoff = 0
ser.rtscts = 0
ser.timeout = 1

ser.open()

#############################
print("FLUSH")
ser.flush()
more = True;
while more:
    bytesIn = ser.read(len(bytesOut))
    print(bytesIn)
    if len(bytesIn) == 0:
        more = False;

#############################
print("ctrlC")
ser.write( ctrlC)
bytesIn = ser.read( 300)
print(bytesIn)
print( len(bytesIn))
replyLength = 107

#############################
if False:
    print("ctrlE")
    ser.write( ctrlE)
    bytesIn = ser.read( 300)
    print(bytesIn)
    print( len(bytesIn))
    replyLength = 5

#############################
print("LOOPS " + str(LOOPS))

for loop in range(1,LOOPS):
    print(loop)

    ser.write( bytesOut)
    ser.write( linebreakOut)

    bytesIn = ser.read( len(bytesOut) + 1)
    print(bytesIn)

    extraIn = ser.read( replyLength)
    print(extraIn)
    print( len(extraIn))

    ####################
    #time.sleep(0.1)

ser.close()

```

dpgeorge, 2021-03-09:
Looks like this may have been fixed in the CODAL in https://github.com/lancaster-university/codal-microbit-v2/issues/80; is that the case?

microbit-carlos, 2021-03-09:
Yes, this should be resolved in CODAL and we'll be able to pull the changes in when we have a new tag.

In the meantime this hex can be used for testing: [microbit-micropython-uart-test-build.hex.zip](https://github.com/microbit-foundation/micropython-microbit-v2/files/6109405/microbit-micropython-uart-test-build.hex.zip)

martinwork, 2021-03-09:
I've run the test twice without any errors, slightly modified on the second go to actually check that the first part of the echo matches.

microbit-carlos, 2021-03-16:
Thanks Martin, Joe and Damien! This fix will be released as part of beta.5.

closed 2021-03-16 by microbit-carlos