microbit-foundation/micropython-microbit-v2 #53: Random number generator is not generating random numbers.

mytechnotalent, 2020-12-26:
```
import random

random_number = random.randint(1, 20)
print(random_number)
```

It is printing the same one random number.  In this case it is 7 @microbit-carlos as we discussed.

carlosperate, 2020-12-26:
uBit.init should be seeding the random number generator:
https://github.com/microbit-foundation/micropython-microbit-v2/blob/0c4287ca33702f85917bd706067a6ca41a21510f/src/codal_app/main.cpp#L52-L53
https://github.com/lancaster-university/codal-microbit-v2/blob/d49c441bb712bc02bf2c7e4d95b757721eb54e4a/model/MicroBit.cpp#L169

dpgeorge, 2021-01-05:
> It is printing the same one random number.

Do you mean it prints the same number each time it starts up?  Calling `randint()` again and again should return different values.

Note that the `random` module uses its own pseudo-RNG, not anything from CODAL.

mytechnotalent, 2021-01-05:
@dpgeorge when I literally reboot the micro:bit it picks the same number every time.  Yes when it starts up.  I was hoping we can avoid having to call `randint()` twice if possible.

dpgeorge, 2021-01-05:
Ok.  I think the issue is that the `random` module uses the same PRNG sequence each time it starts up.  This needs to be fixed so that it seeds itself from a true random source.

mytechnotalent, 2021-01-17:
Any progress on this?  I am working on many micro:python tutorials but the random number generator is failing as stated?

dpgeorge, 2021-01-17:
This should be fixed by b621de7b349d248bdaea4a551fd75d956a1bbe27 (I was not able to test it).

jaustin, 2021-01-20:
@mytechnotalent Does the 'thumbs up' mean it works for you? Just want to be explicit before I close the issue, or go to test it myself :)

mytechnotalent, 2021-01-20:
@jaustin my apologies!  I did `git checkout b621de7` and compile and test...
```
>>> randint(1, 5)
1
>>> randint(1, 5)
4
>>> randint(1, 5)
5
>>> randint(1, 5)
2
>>> 
```

@dpgeorge change did in fact fix.  This is tested and good to merge to main.

closed 2021-01-20 by jaustin