microbit-foundation/micropython-microbit-v2 #54: set_analog_period(20) giving invalid valide

mytechnotalent, 2021-01-03:
As discussed @microbit-carlos:

When setting:
```set_analog_period(20)```

I am getting:
```ValueError: invalid period```

I have tried other values but same result on V2.  12/21 firmware on the v/beta web editor.

microbit-mark, 2021-01-04:
> micro:bit support #42332

Also raised by Paralax on micro:bit support. This works on V1 but not on V2

dpgeorge, 2021-01-05:
Which pin do you set the period on?

Note that the `ValueError: invalid period` error is just passed on from CODAL.

mytechnotalent, 2021-01-05:
@dpgeorge I used `pin0` and `pin1` both together and separately.

mytechnotalent, 2021-01-05:
If they are passed on from CODAL it might be beyond the scope of this repo perhaps @dpgeorge @microbit-carlos?

dpgeorge, 2021-01-11:
The issue is that the CODAL does not allow changing the analog period on pins that are not configured in analog-out mode.  But I think this is how v1 DAL behaved so I don't believe it's an issue with CODAL.

Should be fixed for MicroPython by 6e0273fc4b6b83064912c260f75854f85ab39c6f

jaustin, 2021-01-20:
@mytechnotalent did you test this with the CI build? If so does your thumbs-up mean "works for me"?

mytechnotalent, 2021-01-20:
@jaustin I promise to be more explicit next time my apologies again.  

I `git checkout 6e0273f` and build and flashed and what the `ValueError` is in fact fixed.  Tested and good to merge to main.

```
>>> from microbit import pin1
>>> pin1.set_analog_period(20)
>>> from microbit import pin0
>>> pin0.set_analog_period(20)
>>> 
```

microbit-carlos, 2021-02-17:
Thanks Kevin!
I'll mark this as fixed.

closed 2021-02-17 by microbit-carlos

rbryx, 2021-06-26:
Hi,
I still have this issue. I try to both pin1.set_analog_period(20) and pin1.set_analog_period_microseconds(20) in both Thonny an Mu and still have the error.
My workaround is to pin1.write_analog(0) before trying to pin1.set_analog_period(20).

microbit-carlos, 2021-06-28:
Are you sure you are getting the same error `ValueError: invalid period`? If I flash this programme with the latest version in the Python Editor I do not get any errors:

```
# Add your Python code here. E.g.
from microbit import *

pin1.set_analog_period(20)
```

Could you provide a minimum programme that throws your error? Maybe it's related to some other parts of the code.

rbryx, 2021-06-28:
Sorry, my fault!!!
I was so sure I was using beta-5... three days spent on this... so frustrating!

Again, sorry.

microbit-carlos, 2021-06-28:
No worries, glad you got it working in the end! 👍

Darshnik253, 2022-02-02:
i am making robotic car project in that i have to control dc motor with pwm signal using microbit. so i have written code for that which i combined digital write and analog write pins which part i am attaching below. But code is working only one time after that its not working(error - microbit led is flickering). how can i resolve this error?

pins.digital_write_pin(DigitalPin.P1,0)
pins.digital_write_pin(DigitalPin.P2,0)
pins.digital_write_pin(DigitalPin.P8,1)
pins.digital_write_pin(DigitalPin.P12,0)
pins.analog_write_pin(AnalogPin.P0,511)
pins.analog_set_period(AnalogPin.P0,200)
pins.analog_write_pin(AnalogPin.P3,511)
pins.analog_set_period(AnalogPin.P3,200)

microbit-carlos, 2022-02-02:
Hi @Darshnik253, 
You might be using some pins that are shared with the display (like P3). You can see the pin out here: https://tech.microbit.org/hardware/edgeconnector/#edge-connector-pins

Darshnik253, 2022-02-07:
thank you @microbit-carlos its solve.
but getting another problem in that bluetooth UART service and serial communication using USB if i combined code both of them then after 11 times its not working can u just guide me in that?