microbit-foundation/micropython-microbit-v2 #56: Error message does not scroll line number on V2

microbit-mark, 2021-01-04:
> micro:bit support #42322

> Micro bit v2 will tell you if there is an error eg syntax or name error, but, unlike the original microbit,  does not specify which line the error is on.  It shows a sad face then the error message.  Is there any way, other than using the original microbit, to locate the error?

This is true both of the scrolling message and in the REPL.

V1 contains the line number
V2 does not contain the line number

Could we still offer the line number for V2 to help people locate the error?

carlosperate, 2021-01-04:
Have you tried this in the beta editor? I think it's been included in the beta.3 release of MicroPython.

microbit-mark, 2021-01-04:
Thanks @microbit-carlos. Tested in https://python.microbit.org/v/beta and this is resolved

microbit-carlos, 2021-01-04:
Great, thanks for checking. I'll close this issue then.

closed 2021-01-04 by microbit-carlos