microbit-foundation/micropython-microbit-v2 #57: Pin 9 in display mode V2

microbit-mark, 2021-01-04:
Using `pin9.write_digital(1)` results in `ValueError: Pin 9 in display mode` which we would expect on V1 as the diaply is coupled to the Pin.

Should it be the case as per https://tech.microbit.org/hardware/edgeconnector/#edge-connector-pins that Pin9 is no-longer coupled to the display for V2 and if so can we enable use of pin9 without having to disable the display?

```
MicroPython v1.13 on 2020-12-21; micro:bit v2.0.0-beta.3 with nRF52833
Type "help()" for more information.
>>> from microbit import *
>>> pin9.write_digital(1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Pin 9 in display mode
```

jaustin, 2021-02-12:
See https://github.com/microbit-foundation/micropython-microbit-v2/pull/64 - will test once there's a CI build

dpgeorge, 2021-02-17:
Fixed by a94e53de7e7d6856a97b71ad5667501b05e7bf03

closed 2021-02-17 by dpgeorge