microbit-foundation/micropython-microbit-v2 #59: Error displays incorrect line

microbit-mark, 2021-01-19:
> micro:bit support: #42322

In this example, a syntax error is introduces on line 3, removing the `:` from the loop

```Python
from microbit import *

while True
    display.scroll(str(temperature()))
    display.scroll(' celsius ')
```

However, The micro:bit displays that there is a syntax error on line 4. This is the same in the REPL console.

It should display line 3.

The ticket requester states that this does not reproduce in Mu though, which I expect means that it does not reproduce for V1 boards

microbit-carlos, 2021-01-19:
Just to add a bit more info, I tried this in a V1 and the error report is the same at line 4.
The comment about Mu might be from the editor error checker:
[image]

Not sure to what extent I'd consider this a bug to fix, it'd be an upstream change.

jaustin, 2021-01-20:
@dpgeorge do you have any plans to change this upstream? Otherwise I think we should close for now as 'v1/v2 behave the same'

dpgeorge, 2021-02-14:
This is a generic "feature" of MicroPython that is in all ports.  There are no plans to change it, I don't think it's that confusing.

microbit-carlos, 2021-02-15:
Thanks Damien, we'll close this as something that won't be changed.

closed 2021-02-15 by microbit-carlos