microbit-foundation/micropython-microbit-v2 #60: Feature Request: programmable power managment

microbit-mark, 2021-01-26:
**Is your feature request related to a problem? Please describe.**

User story: I need to make a sensing device that lasts a long time on batteries. Thus the device needs to power off under program control, and power up on an event, take a measurement, then go to sleep again.

**Describe the solution you'd like**
CODAL offers a power management API https://github.com/lancaster-university/microbit-v2-samples/blob/master/source/samples/PowerManagementTest.cpp and there is a published spec https://github.com/microbit-foundation/spec-power-management. It would be useful to surface this in MakeCode

**Describe alternatives you've considered**
None

**Additional context**
Add any other context or screenshots about the feature request here.

dpgeorge, 2022-08-21:
Initial `power` module has been implemented in 009fbd28c6c7e2e2f03ccd622e5087f05177a144.

It currently does not support the `run_every` argument to `deep_sleep()`.  But everything else is there.

I noticed that `deep_sleep()` doesn't work very well.  Some issues I noticed are:
- if `ms` and `buttons` are specified then only `ms` works (buttons do not wake it)-
- specifying `buttons=button_b` also allows waking it by button A
- the first time `deep_sleep(1000)` is called, the microphone LED comes on after it wakes

dpgeorge, 2022-08-24:
The `pins` and `buttons` arguments have been merged together into the `wake_on` argument, see f0fe9823ebd428569acb1ebb2fa7be6972885c48

dpgeorge, 2022-08-24:
The `run_every` parameter is implemented by 2233d79b8ab15bb769adaa44d53874ce80ebd3eb

Due to the issue with CODAL that a button/pin won't stop the timeout, `run_every` doesn't work as expected if `wake_on` is specified (the wake source will not wake the device, it will only wake when the next `run_every` event is scheduled).

microbit-carlos, 2022-08-25:
> Due to the issue with CODAL that a button/pin won't stop the timeout, `run_every` doesn't work as expected if `wake_on` is specified (the wake source will not wake the device, it will only wake when the next `run_every` event is scheduled).

Just to clarify, I think the wake source does wake up the target, but the function `power.deep_sleep()` is blocking until the "sleep time" elapses:
- https://github.com/lancaster-university/codal-microbit-v2/issues/221

With this Python script, it goes to sleep when pressing A, and then when pressing B it wakes up showing the letter `S` in the display (which was set right before going to sleep). It then waits there until the 10 seconds of sleep elapse, and only at that point it shows the `!` in the display and continues running:

```python
from microbit import *
import power

@run_every(s=30)
def foo():
    display.show(Image.SURPRISED)
    sleep(150)

while True:
    display.show(Image.HAPPY)
    if button_a.is_pressed():
        display.show("S")
        sleep(100)
        power.deep_sleep(10000, wake_on=button_b, run_every=True)
        display.show("!")
        sleep(500)
    sleep(200)
```

dpgeorge, 2022-08-30:
> Just to clarify, I think the wake source does wake up the target, but the function `power.deep_sleep()` is blocking until the "sleep time" elapses

Yes, I agree, that looks to be the behaviour.

For MicroPython, I think we need a `uBit.power.deepSleepAtMost(ms)` (or similar) API.  Essentially a non-blocking deep sleep.  Does something like that already exist?

microbit-carlos, 2022-09-05:
We've shipped this in it's current state in v2.1.0-beta.1, so we can close this as part of that milestone, and any additional changes captured in new tickets.

closed 2022-09-05 by microbit-carlos