microbit-foundation/micropython-microbit-v2 #65: read_analog() errors if script reads multiple pins -V2 only

AndyLindsay, 2021-02-16:
If a script calls read_analog() for more than one pin in a loop, incorrect measurements are returned periodically.  The errors were evident in micro:bit V2, but not micro:bit V1.  Micro:bit Education Foundation Tech Support noticed an issue that might be related in [lancaster-university
/ codal-microbit-v2 / Erratic output when reading multiple analog pins #81](https://github.com/lancaster-university/codal-microbit-v2/issues/81).  

**Test Script**
```

# analog_read_issue_test

# connect pin0 to 3.3 V and pin2 to 0 GND.

from microbit import *

while True:
v0 = pin0.read_analog()
v2 = pin2.read_analog()
print("v0 =", v0, "v2 =", v2)
if v0 < 1010:
        print("pin2 measurement error")
        sleep(2000)
    if v2 > 10:
        print("pin0 measurement error")
        sleep(2000)
    sleep(100)

```
**Sample from Terminal Output**

v0 = 1020 v2 = 0
v0 = 1022 v2 = 0
v0 = 0 v2 = 0
pin2 measurement error
v0 = 1021 v2 = 1
v0 = 1020 v2 = 1
   ...
v0 = 1020 v2 = 0
v0 = 1021 v2 = 1021
pin0 measurement error
v0 = 1020 v2 = 1
v0 = 1021 v2 = 1
v0 = 1021 v2 = 0

**Full Speed Test** 
```

# analog_read_issue_test_full_speed

# connect pin0 to 3.3 V and pin2 to 0 GND.

from microbit import *

while True:
    v0 = pin0.read_analog()
    v2 = pin2.read_analog()
    if v0 < 1010:
        print("v0 =", v0, "v2 =", v2)
        print("pin2 measurement error")
        sleep(2000)
    if v2 > 10:
        print("v0 =", v0, "v2 =", v2)
        print("pin0 measurement error")
        sleep(2000)
    #sleep(100)

```
**Samples from Terminal Output**

v0 = 1 v2 = 0
pin2 measurement error
v0 = 1019 v2 = 1021
pin0 measurement error
v0 = 1021 v2 = 1019
pin0 measurement error
v0 = 1022 v2 = 1020

microbit-mark, 2021-02-17:
@martinwork could you take a look at this and see if it is related to https://github.com/lancaster-university/codal-microbit-v2/issues/81 or something else?

martinwork, 2021-02-17:
It appears to be the same problem. I'm experimenting with a fix in CODAL right now.

microbit-carlos, 2021-03-09:
@martinwork do you know what is the current state of this in CODAL?

martinwork, 2021-03-09:
CODAL has issue https://github.com/lancaster-university/codal-microbit-v2/issues/81. I have made an attempt at Joe's idea for a solution, in PR https://github.com/lancaster-university/codal-nrf52/pull/24, but it has a problem reading the DMA buffers from user code. https://github.com/lancaster-university/codal-microbit-v2/issues/81#issuecomment-783367236.

microbit-carlos, 2021-06-02:
@martinwork has this issue been resolved with https://github.com/lancaster-university/codal-nrf52/pull/30?

If so, could you check the latest CI MicroPython build to confirm the original code listed here works correctly now?

martinwork, 2021-06-02:
@microbit-carlos Yes, with MicroPython Version: 1.0.1/2.0.0-beta.5, I see the swapped values very easily. With the latest CI build I don't see those errors. I did see the above code flag occasional values as errors, where the values exceed the limits checked in the code.

Running the code below for a few minutes, I saw minimum v0 = 981, maximum v2 = 33, but switching from crocodile leads to a pinbetween and jumper wires, changed that to minimum v0 = 1018, maximum v2 = 3.

```
from microbit import *

v0min = 1023
v2max = 0

while True:
    v0 = pin0.read_analog()
    v2 = pin2.read_analog()
    if v0min > v0:
        v0min = v0
    if v2max < v2:
        v2max = v2
    print("v0 =", v0, "v2 =", v2, "v0min =", v0min, "v2max =", v2max)
```

microbit-carlos, 2021-06-07:
Thanks @martinwork! I think the values that you've seen are expected given the cabling, so I'll close this as resolved.

Thanks @AndyLindsay for the report as well!

closed 2021-06-07 by microbit-carlos