microbit-foundation/micropython-microbit-v2 #68: backward compatibility V1-> v2

JLFra, 2021-03-16:
I don't understand why you haven't achieved backward compatibility between V1 and V2. So programs that worked perfectly in V1 no longer work in V2, it's incomprehensible. Why didn't you do what was necessary? For example, I used the blokcly talky extension in BLE which works fine in V1, and nothing works in V2. I admit that I do not understand because everything worked perfectly in V1 and there it is complicated. People have spent a lot of time creating extensions, with the switch to V2 nothing works anymore. For example the sampling of an analog voltage is badly sampled with non-constant time at start-up? In addition, the Android application does not work at all. I even have a V2 card which is totally crashed with the display of an icon followed by a number which is constantly displayed, even after flashing the V2 firmware. In short, very complicated to use your V2 which poses big concerns. Nothing to do with V1, which is very good. I don't understand why you didn't develop reliable firmware before it came to the market. Thank you for your quick help and for making V2 reliable. Why your V2 is still not compatible with the Mu IDE, while the V1 works perfectly with MU. When will you finally offer the standard BLE python commands now that you have to increase the memory. Personally, I do not see the interest of your V2 which brings nothing and even less than the V1. Thank you for your very precise answers and your quick help.

IdarinC, 2021-03-17:
What issues are you having with Mu? Have you loaded the latest hex? https://github.com/microbit-foundation/micropython-microbit-v2/releases/tag/v2.0.0-beta.5  It works without issue for me using Mu 1.0.3 and 1.1.0 beta 2

JLFra, 2021-03-17:
I had put the microbit V2under miantenance mode, and drag and drop the file hex into V2. I had start Mu 1.1.0 alpha2. select microbit. But no repl, impossible to flash. Do you have the right procedure please.
But in general, there is to many issues with V2, nothing has been test correctly. So V1 is much better.
I thought that with the new card and its new hardware and its larger memory, that this would bring us real progress. It's quite the opposite. The developed firmware does nothing. We spend too much time debugging the system. Whereas when V1 was released, everything worked the first time. There it is a real disaster. Nothing with BLE, sampling analog signals is a concern, no fft for sound analysis. In short, the V2 only provides a simple microphone and very basic speaker with which you can hardly do anything. More V1-> V2 compatibility. Shame. And to think that we will soon find no more V1 and only V2. Really too bad. This V2 card was released in Nov 2020, and since then, nothing new, nothing has progressed.

On the official next page nothing about the Mu IDE, only the microbit python IDE which is not suitable at all, no autocompletion, no REPL, nothing to do with MU. Why Microbit does not comminute on MU which is really much better without comparison with the proprietary tool of Microbit.
https://microbit.org/fr/get-started/first-steps/set-up/
Why is there no explanation for using Mu on the official page for using the Mu IDE?

In short, everything is very complicated with V2

There is plubicity with the application on Android, but nothing works. Why all the publicity when nothing is working?
I admit I am extremely disappointed with this V2, while I am very reliant on the V1.

microbit-mark, 2021-03-19:
Maintaining software compatibility between all micro:bit revisions is of great importance and you should be able to [use the latest V2 micro:bit in the same way you would the original](https://support.microbit.org/support/solutions/articles/19000119164-will-the-original-micro-bit-work-with-the-latest-micro-bit-).  If you have an issue with a particular program, you may need to [update the .hex file](https://support.microbit.org/support/solutions/articles/19000090748-updating-your-hex-file-) in the editor to use it with all micro:bits.

If you see any errors, this knowledgebase article [describes the errors and what you need to do to resolve them](https://support.microbit.org/support/solutions/articles/19000016969-micro-bit-error-codes).

The Python Editor at https://python.microbit.org supports all micro:bit revisions and as of this week, so does the latest release of the Mu editor. 🎉 

You can read about the [Mu release to support V2](https://madewith.mu/mu/users/2021/03/15/beta2.html) and download it from the Mu website https://codewith.mu/en/download

Features such as the [REPL](https://python-editor-2-1-1.microbit.org/help.html?snippets=true#REPL) are available in the current editor, providing that the browser you are using [supports WebUSB](https://support.microbit.org/support/solutions/articles/19000084059-direct-flashing-from-the-browser-with-the-apps-and-editors).

The Python Editor beta https://python.microbit.org/v/beta also contains the [autocomplete feature](https://support.microbit.org/support/solutions/articles/19000098541-beta-testing-autocomplete-for-the-python-editor).
 
The micro:bit Android app is also open source and you will see [Android app issues](https://github.com/microbit-foundation/microbit-android/issues)  that the developers are aware of in the repository. You can find support for the [pairing and flashing process](https://support.microbit.org/support/solutions/articles/19000065804-how-do-i-pair-my-bbc-micro-bit-using-the-android-app-) in the knowledgebase and you can also sign up to test the [android beta via the play store](https://play.google.com/store/apps/details?id=com.samsung.microbit).

I hope that resolves the concerns raised here.

indigo51, 2023-07-14:
Connect a simple Grove kit button sensor from the BitGadget kit and load this Python on V2 microbit:

from microbit import *
while True:
    reading = pin0.read_analog()
    print(reading)
    sleep(500)

The BitMaker board has a power switch that supplies either 5V or 3.3V to attached sensors. 

**SIMPLE BUTTON SWITCH pressed on**
**Using a V2 microbit:**
When the button is ON the reading is 1020 with BitMaker board is set to 3.3V (CORRECT)
and 789 when it is set to 5V. (VERY INCORRECT)

**Using a V1 microbit:**
When the button sensor is on the reading is 1020 with BitMaker board set to 3.3V  (CORRECT!)
and 1023 when it is set to 5V.(CORRECT!)

**SOUND SENSOR testing max sound input**
**Using a V2 microbit:**
Sound sensor max reading is 900 with BitMaker board set to 3.3V (INCORRECT)
and 1023 when it is set to 5V. (CORRECT)

**Using a V1 microbit:**
Sound sensor max reading is 1023 with BitMaker board set to 3.3V  (CORRECT!)
and 1023 when it is set to 5V.(CORRECT!)

In other words, the V1 microbit seems to handle both 5V and 3.3V sensors perfectly, whereas V2 does not.

Any ideas?

martinwork, 2023-07-18:
@indigo51 I am not familiar with the BitGadget kit, and I failed to find good documentation online, but note that connecting 5V to the micro:bit edge connector of V1 or V2 may damage micro:bit. I suspect the 5V option is intended for use with particular sensors or devices. I'm pretty sure this isn't a Python issue. If you need more information, feel free to create a support ticket at https://support.microbit.org/support/home.