microbit-foundation/micropython-microbit-v2 #69: With V2, music.play(pin=pin1) permanently overrides pin0 default

martinwork, 2021-03-22:
Testing with Editor Version: 2.2.0-beta.5; MicroPython Version: 1.0.1/2.0.0-beta.5

With V1 specifying the pin only affects the current instruction.
With V2 subsequent instructions use pin1. It's possible to change to pin2, and back to pin1, but not pin0.

With V2, the code below stops looping, usually within a minute or so.

```
import microbit
from microbit import *
import music

while True:
    microbit.display.scroll('0')
    music.play(music.JUMP_UP)
    sleep(1000)
    
    microbit.display.scroll('1')
    music.play(music.JUMP_UP, pin=pin1)
    sleep(1000)
    
    microbit.display.scroll('0')
    music.play(music.JUMP_UP)
    sleep(1000)

    microbit.display.scroll('0')
    music.play(music.JUMP_UP, pin=pin0)
    sleep(1000)

```

dpgeorge, 2021-05-03:
I can reproduce the lock up with the following modified version of the above test:
```python
from microbit import *
import music

set_volume(5)

while True:
    microbit.display.show('0')
    music.play(music.JUMP_UP)
    microbit.display.show('1')
    music.play(music.JUMP_UP, pin=pin1)
    microbit.display.show('0')
    music.play(music.JUMP_UP)
    microbit.display.show('0')
    music.play(music.JUMP_UP, pin=pin0)
```

dpgeorge, 2021-05-24:
The lock up bug is fixed by 291e05c8440de4e90dcf1f443136d2b7005272a1 (at least, running the above test, it doesn't lock up anymore).

Still need to look into the overriding of pin0 issue.

dpgeorge, 2021-06-07:
The issue overriding pin0 seems to be a bug in CODAL.  Here is a minimal reproduction in C++:
```c
#include "MicroBit.h"

MicroBit uBit;

int main() {
    uBit.init();

    uBit.serial.printf("== start ==\r\n");
    uBit.serial.printf("P0.name=%d\r\n", uBit.io.P0.name);
    uBit.serial.printf("P1.name=%d\r\n", uBit.io.P1.name);
    uBit.serial.printf("P2.name=%d\r\n", uBit.io.P2.name);

    uBit.audio.setPin(uBit.io.P1);

    uBit.serial.printf("== end ==\r\n");
    uBit.serial.printf("P0.name=%d\r\n", uBit.io.P0.name);
    uBit.serial.printf("P1.name=%d\r\n", uBit.io.P1.name);
    uBit.serial.printf("P2.name=%d\r\n", uBit.io.P2.name);

    for(;;);
}
```
Output (using v0.2.28 of CODAL):
```
== start ==
P0.name=2
P1.name=3
P2.name=4
== end ==
P0.name=3
P1.name=3
P2.name=4
```
Note that P0 has turned into P1!

dpgeorge, 2021-06-08:
Pin override issue fixed in CODAL in https://github.com/lancaster-university/codal-microbit-v2/commit/ab3f8b3522c909c57f83fc40432b1ff5883e251e

When there's a new release of CODAL that includes this commit it can be updated here, and this issue closed.

dpgeorge, 2021-06-21:
Updated to latest CODAL in ab59d87070e97c7c89a986947f7c568660834029, so this issue should be fully resolved.

closed 2021-06-21 by dpgeorge