microbit-foundation/micropython-microbit-v2 #70: UART does not return to REPL

microbit-mark, 2021-03-24:
```
from microbit import *
def uart_test():
    uart.init(baudrate=115200,tx=pin0,rx=pin1)
    display.scroll("UART")
    uart.init(baudrate=115200)
    return ('done son')
```

On v1 program displays 'UART' and returns the string 'done son' in the REPL

on V2 program displays 'UART' but does not return the string 'done son' in the REPL

Tested in https://python.microbit.org/v/beta

Clicking on on send CTRC-C or CTRL-D does not reset the REPL

microbit-carlos, 2021-03-24:
Removing the `display.scroll()` and the function shows the same issue:

```python
from microbit import *

print("this prints normally")
sleep(1000)
uart.init(baudrate=115200,tx=pin0,rx=pin1)
print("this goes to the edge connector pins")
sleep(1000)
uart.init(baudrate=115200)
print("this doesn't print on serial")
sleep(1000)
```

I can replicate this as well in MakeCode (the stable deployment with CODAL 0.2.25:

```typescript
serial.writeLine("this prints normally")
basic.pause(1000)
serial.redirect(SerialPin.P0, SerialPin.P1, BaudRate.BaudRate115200)
serial.writeLine("this goes to the edge connector pins")
basic.pause(1000)
serial.redirect(SerialPin.USB_TX, SerialPin.USB_RX, BaudRate.BaudRate115200)
serial.writeLine("this doesn't print on serial")
basic.pause(1000)
```

@martinwork could you try to create a test in C++ to see if this is reproducible in the CODAL side?

If a normal call to `uBit.serial.redirect()` with the edge connector pins, and then with the internal pins doesn't work we can try to replicate the MicroPython uart.init call:

https://github.com/microbit-foundation/micropython-microbit-v2/blob/e0f3e60c5e7755eb71c789b3a765d271320b2e17/src/codal_port/microbit_uart.c#L86
https://github.com/microbit-foundation/micropython-microbit-v2/blob/b7a76e449cf1b437972c21efaf468fbe5891c6a2/src/codal_app/microbithal.cpp#L210-L221

martinwork, 2021-03-25:
Created https://github.com/lancaster-university/codal-microbit-v2/issues/92

microbit-carlos, 2021-04-26:
@martinwork thanks for looking into this!
Is PR https://github.com/lancaster-university/codal-nrf52/pull/31 the only thing pending for this being fixed in CODAL, or are there more?

martinwork, 2021-04-27:
@microbit-carlos I believe that PR on its own should fix this issue, but I haven't tested it in a micropython build.

dpgeorge, 2021-05-03:
This is now waiting on a tag in codal-microbit-v2

microbit-mark, 2021-05-21:
Latest tagged release is in https://github.com/lancaster-university/codal-microbit-v2/releases/tag/v0.2.26 which I think should resolve this issue.

dpgeorge, 2021-05-24:
Fixed by 291e05c8440de4e90dcf1f443136d2b7005272a1

closed 2021-05-24 by dpgeorge