microbit-foundation/micropython-microbit-v2 #72: Need library framebuf

crouse12, 2021-04-02:
ESP32 and Esp8266 have a library, called framebuf.
Thus, they can show string or draw a picture on oled or Max7219.
However, I found that microbit v2 does not have this library.
Can you add it?

microbit-carlos, 2021-04-26:
Hi @crouse12,

You can build a version of MicroPython with the framebuf module included. Hopefully this steps should be relatively easy:

- Clone this repository and follow all the build steps from the README
- Add `#define MICROPY_PY_FRAMEBUF (1)` somewhere inside `src/codal_port/mpconfigport.h`
    - This area is where all the other modules are configured:
       https://github.com/microbit-foundation/micropython-microbit-v2/blob/64c722c4001864579296ec42f26609dfabb8a92a/src/codal_port/mpconfigport.h#L81-L89
- Rebuild again, and the `framebuf` module should now be available  🎉

crouse12, 2021-04-27:
Could you tell me how to put this new version of micropython into pythoneditor?
I need to generate hex file for microbit v2.

microbit-carlos, 2021-04-27:
You can replace the `microbit-micropython-v2.hex` file in https://github.com/bbcmicrobit/PythonEditor/tree/master/micropython.

You can also send files to the micro:bit from your computer with https://github.com/ntoll/microfs.

microbit-carlos, 2021-06-01:
I'll close this issue as resolved, but feel free to reopen if you have any questions in this area!

closed 2021-06-01 by microbit-carlos