microbit-foundation/micropython-microbit-v2 #73: Random freezes after music.pitch with beta 5

bertbaron, 2021-04-02:
I experience random freezes of a program on the micro:bit V2 with beta 5. I wasn't able to reproduce this with beta 4.
Actually I started observing those freezes after flashing a program earlier this week with the online python editor. I didn't have any problems with approximately the same program flashed using the same editor a few weeks earlier. I'm not sure which micropython is used by that editor but it seems related.

With the following minimized program the issue can be reproduced:
[code]
from microbit import *
import time
import music

music.pitch(220, 100)

blink_on = False
while True:
    display.show(Image.HEART if blink_on else Image())
    blink_on = not blink_on
    time.sleep(0.1)
[/code]

It may require some time or a few attempts (resets) to reproduce but should be possible within about 30 minutes.

Without the music.pitch I wasn't able to reproduce so far, so I think its related.

dpgeorge, 2021-05-03:
Thanks for the report, I can reproduce the issue by running the above script many times (with just a soft reset between runs); eventually it stops blinking the heart image.

The following simpler version also seems to trigger the same bug (it has the same symptoms):
```python
from microbit import *
import music

speaker.off()

while True:
    music.pitch(220, 5)
    display.show(Image())
    sleep(2)
    display.show(Image.HEART)
    sleep(2)
```
Just run that once and it should eventually stop blinking the heart image.

This issue could be related to #69.

dpgeorge, 2021-05-03:
Simpler version that still locks up:
```python
from microbit import *
import music
    
speaker.off()
        
while True:
    music.pitch(220, 5, wait=0)
    display.show(Image.HEART)
    display.show(Image())
```

wodnjs6512, 2021-05-06:
im experiencing similar issue

let microbit to receive UART for execution and using music library may occur the random freeze, as stated by @dpgeorge @bertbaron 

btw is there a way to get beta4 firmware for check purpose? 

All I could get from official website was beta5...

dpgeorge, 2021-05-11:
I have managed to reproduce this in pure C++ using just the CODAL.  Run the following program and eventually it will stop (the LEDs stop flickering).
```c
#include "MicroBit.h"

const char * const img_data_blank ="\
    000,000,000,000,000\n\
    000,000,000,000,000\n\
    000,000,000,000,000\n\
    000,000,000,000,000\n\
    000,000,000,000,000\n";

const char * const img_data_heart ="\
    000,255,000,255,000\n\
    255,255,255,255,255\n\
    255,255,255,255,255\n\
    000,255,255,255,000\n\
    000,000,255,000,000\n";

MicroBit uBit;

void gesture_event_handler(Event evt) { 
}

int main() { 
    MicroBitImage img_blank(img_data_blank);
    MicroBitImage img_heart(img_data_heart);

    uBit.init();
    uBit.audio.setSpeakerEnabled(false);

    // This is needed
    uBit.messageBus.listen(DEVICE_ID_GESTURE, DEVICE_EVT_ANY, gesture_event_handler);

    for (;;) { 
        // Either of these (or both together) will lead to the lock-up
        uBit.audio.virtualOutputPin.setAnalogValue(128);
        uBit.audio.virtualOutputPin.setAnalogPeriodUs(4545);

        uBit.display.print(img_heart);
        uBit.display.print(img_blank);

        // This is needed
        Event(DEVICE_ID_SCHEDULER, DEVICE_SCHEDULER_EVT_IDLE);
    } 
}
```
@finneyj do you have any idea why the above C++ code will eventually lock up?

finneyj, 2021-05-11:
Thanks @dpgeorge @bertbaron . We might be tracking the same bug over here too I think:
https://github.com/lancaster-university/codal-microbit-v2/issues/102

Just reading your comments in the code, can I assume that the bug does not repro if the `Event(DEVICE_ID_SCHEDULER, DEVICE_SCHEDULER_EVT_IDLE);` and associated handler are removed?

microbit-carlos, 2021-05-11:
v0.2.24: https://github.com/microbit-foundation/micropython-microbit-v2/blob/v2.0.0-beta.4/src/codal_app/codal.json#L5

finneyj, 2021-05-11:
Just ran caught this in a hardware debugger. It seems to be stuck in an I2C transaction with the accelerometer. This is consistent with what we saw on https://github.com/lancaster-university/codal-microbit-v2/issues/102...

I think you can leave this one with me @dpgeorge!

[image]

finneyj, 2021-05-11:
Hi @dpgeorge @microbit-carlos @bertbaron 

Just to report back, I suspect we're actually looking at a silicon bug here. Not sure why this hasn't shown up previously though...

See here for tracking: https://github.com/lancaster-university/codal-microbit-v2/issues/102
And here for a PR that addresses the repro case above: https://github.com/lancaster-university/codal-nrf52/pull/36

If all goes well, this will be merged into the next CODAL release (v0.2.26)

microbit-carlos, 2021-05-21:
@finneyj have you had a chance to report the bug to the Nordic Dev Zone? If so, have they replied back with any additional info?

finneyj, 2021-05-21:
no, sorry... the patch is in the codal-microbit-v2 v0.2.26 tag though.

dpgeorge, 2021-05-24:
Fixed by 291e05c8440de4e90dcf1f443136d2b7005272a1

closed 2021-05-24 by dpgeorge