microbit-foundation/micropython-microbit-v2 #74: Uart problem

crouse12, 2021-04-04:
On v2, uart never return radio receiving message.

[image]

Tested in https://python.microbit.org/v/2
In order to test above code, I use another microbit v2 to send radio message and use python serial to read uart message.

send radio code
[image]

python code to read uart message
[image]

martinwork, 2021-04-04:
I tried the code below in a V1 and a V2. It seems to work with V2 as radio receiver of sender.

I initially received the serial messages with tera term, then used the python below.

micro:bit code
```
from microbit import *
import radio

radio.on()

uart.write('Hello'+'\n')

while True:
    
    display.set_pixel( 0, 0, 0 if display.get_pixel(0,0) else 9)
    
    incoming = radio.receive()
    if incoming != None:
        uart.write(incoming + '\n')
        display.scroll('R')
        
    if button_a.was_pressed():
        radio.send('flash\n')
        display.scroll('F')
```

Python to receive serial messages
```
import subprocess
import shutil
import os
import stat
import distutils.dir_util
import filecmp
import json
import sys
import time

import serial
import serial.tools
import serial.tools.list_ports

PORT=''
#PORT='COM5'

BAUDRATE = 115200
#BAUDRATE = 57600

port = PORT

if len(port) == 0:
  listPortInfo = serial.tools.list_ports.comports(include_links=False)
  port = listPortInfo[0].device

print("PORT " + port)

ser = serial.Serial()
ser.port = port
ser.baudrate = BAUDRATE
ser.bytesize = 8
ser.parity = 'N'
ser.stopbits = 1
ser.timeout = 1
ser.xonxoff = 0
ser.rtscts = 0

ser.open()

ser.flush()

try:
    while True:
        bytesIn = ser.readline()
        print(bytesIn)
        time.sleep(0.1)

except KeyboardInterrupt:
    ser.close()
    print('Close')

```

crouse12, 2021-04-04:
Sorry! I give wrong python code.
radio message disappeared if you call uart.write quickly.

[image]

martinwork, 2021-04-05:
Here are the 3 programs, pasted below as text and here as files rather than PNG, so they are easy try out: [issue74.zip](https://github.com/microbit-foundation/micropython-microbit-v2/files/6258013/issue74.zip)

I have modified the desktop python script to find the port number, and so that it doesn't print "Hello\n", to make it easier to see "flash\n\n".

If these programs aren't right, please send correct programs.

I can't see what's wrong. What am I missing? Please explain further what the problem is.

Also I tried a dual send/receive program that counts radio messages, and uses `is_pressed ` rather than `was_pressed`. This shows that radio messages sent quickly get lost, but it seems to be similar on V1 and V2 and only slightly changed by uart.write. When using `was_pressed` as the trigger I don't see any lost messages.

Receive
```
import radio
from microbit import *
radio.on()
while True:
    uart.write('Hello'+'\n')
    incoming = radio.receive()
    if incoming != None:
        uart.write(incoming+'\n')

```

Transmit
```
import radio
from microbit import *
radio.on()
while True:
    if button_a.was_pressed():
        radio.send('flash\n')

```

Desktop
```
import serial
import serial.tools
import serial.tools.list_ports

PORT=''
#PORT='COM5'

BAUDRATE = 115200

port = PORT

if len(port) == 0:
  listPortInfo = serial.tools.list_ports.comports(include_links=False)
  port = listPortInfo[0].device

ser = serial.Serial(port, BAUDRATE)

normal = b'Hello\n'

try:
    while True:
        data_raw = ser.readline()
        if data_raw != normal:
            print(data_raw)

except KeyboardInterrupt:
    ser.close
    print('Close')

```

Count radio messages
```
import radio
from microbit import *

count_receive = 0
count_send = 0

radio.on()

while True:
    uart.write('Hello\n')
    
    incoming = radio.receive()
    if incoming != None:
        count_receive = count_receive + 1
        uart.write(str(count_receive)+" "+incoming+'\n')
        
    if button_a.is_pressed():
        count_send = count_send + 1
        radio.send(str(count_send))

```

crouse12, 2021-04-05:
If you only print radio message, radio message will not disappear.
But, I need to print uart Hello and radio message at the same time.
The files are shown as follows.
https://drive.google.com/file/d/1aULYjuRjPwa_NaR7A3Rt_R8U-Hhjm1So/view?usp=sharing
That is because I try to create some scratch 3.0 blocks for microbit now.
https://www.youtube.com/watch?v=ywIxgYDro7A
Thus, I find that uart message will make radio message disappear.
I also find a new bug for uart, accelerometer and radio, since I use my scratch blocks.
However, I need time to find what happened.

martinwork, 2021-04-05:
Thanks. Those files appear to match the ones I typed in. When I test those files, the text "flash" is appearing in the output of the python program between all the "Hello"s, but scrolls off the screen very quickly.

In a Windows Command Prompt, I used the command "python 55.py >pyout.txt" to pipe the python print output into a text file, and pressed the radio sending micro:bit's A button 10 times, before using Ctrl-C to exit the python program. Whether I test with V1s or V2s the pyout.txt file has 10 occurrences of "flash" and several thousand "Hello"s.

microbit-carlos, 2021-04-26:
Thanks for all the debugging Martin!

@crouse12 did you manage to resolve the issue you were having?

crouse12, 2021-04-27:
> Thanks for all the debugging Martin!
> 
> @crouse12 did you manage to resolve the issue you were having?

YES

microbit-carlos, 2021-04-27:
Great, in that case I'll close this issue. Thanks for the report and the update, and thanks again Martin for the debugging!

closed 2021-04-27 by microbit-carlos