microbit-foundation/micropython-microbit-v2 #77: Bluetooth: a hex created with editor 2.2.0 does not include the partial flashing service

martinwork, 2021-05-10:
[image]

[microbit_program (9).zip](https://github.com/microbit-foundation/micropython-microbit-v2/files/6454217/microbit_program.9.zip)

microbit-carlos, 2021-05-24:
How can we verify if the partial flashing is present or not?
As far as I understood it should be there simply by using CODAL?

martinwork, 2021-05-24:
> How can we verify if the partial flashing is present or not?

Check with nRF Connect or similar.

> As far as I understood it should be there simply by using CODAL?

This is required for the partial flashing service:

`"MICROBIT_BLE_PARTIAL_FLASHING" : 1,
`

These are the relevant settings:

```
"MICROBIT_BLE_ENABLED" : 0,
"MICROBIT_BLE_PAIRING_MODE": 1,
"MICROBIT_BLE_DFU_SERVICE": 1,
"MICROBIT_BLE_DEVICE_INFORMATION_SERVICE": 1,
"MICROBIT_BLE_EVENT_SERVICE" : 0,
"MICROBIT_BLE_PARTIAL_FLASHING" : 1,
"MICROBIT_BLE_SECURITY_LEVEL": "SECURITY_MODE_ENCRYPTION_NO_MITM"
```

I think the above are the ones micropython needs, and are all defaults except MICROBIT_BLE_PARTIAL_FLASHING and MICROBIT_BLE_SECURITY_LEVEL/MICROBIT_BLE_SECURITY_MODE 

MICROBIT_BLE_SECURITY_MODE 2 is the same as SECURITY_MODE_ENCRYPTION_NO_MITM

MICROBIT_BLE_EVENT_SERVICE and MICROBIT_BLE_PARTIAL_FLASHING default to zero because no default is defined in MicroBitConfig.h.

microbit-carlos, 2021-05-25:
Gave that a spin and got this additional service, I assume that means it worked?

[image]

jaustin, 2021-05-25:
@martinwork is this an issue of the config MicroPython uses to build? If so could you please suggest what @dpgeorge should add?

martinwork, 2021-05-25:
Yes. It's the micropython build that the editor uses that is lacking the PF service.

"MICROBIT_BLE_PARTIAL_FLASHING" : 1 makes the build include the PF service

dpgeorge, 2021-05-28:
Added in f0e630bb0b03d8c22ec98d131ee63cc0a2524bb9

closed 2021-05-28 by dpgeorge

microbit-carlos, 2021-06-01:
Thanks Damien!
@martinwork could you check if you can use partial flashing with the Hex built via CI (download in the Artifacts section)? https://github.com/microbit-foundation/micropython-microbit-v2/actions/runs/887027270

martinwork, 2021-06-01:
@microbit-carlos Yes, partial flashing works with that hex. The current iOS app beta has code which pre-dates the switch to using a crc of the version string, but it works if I remove that check. I'll update the app.