microbit-foundation/micropython-microbit-v2 #79: Micro:Bit freeze when analog read

ardaij, 2021-06-05:
Hi,

In our school, we programming micro:bits in MicroPython. For analog reading, the microbit freezes at irregular intervals.

The error can be reproduced with this code:

`from microbit import *

light = False

while True:
pin1.read_analog()
display.set_pixel(2, 2, 5 if light else 0)
light = not light
sleep(200)
`

We use microbit v2 microcontrollers: MicroPython v1.13 on 2021-03-16; micro:bit v2.0.0-beta.5 with nRF52833
We tested this with 7 of the same Micro:Bits.

Please help us.

I also asked this question in another forum because I don’t know where we can get help.

Thanks

dpgeorge, 2021-06-05:
This is likely the same as #73, and it was fixed by commit 291e05c8440de4e90dcf1f443136d2b7005272a1

Testing the above script with that commit, it doesn't freeze.

ardaij, 2021-06-05:
I just built the micropython from the github repository, and I copied the hex file onto the microbits.

MicroPython v1.15-64-g1e2f0d280 on 2021-06-05; micro:bit v2.0.0-beta.5 with nRF52833

I ran the program for several hours without error. This seems to have solved the problem.

Thanks

And thank you for the MicroPython...

microbit-carlos, 2021-06-08:
Glad you got it working and thanks for the report @ardaij!
I'll close this as resolved as the fix will be present in the next release.

closed 2021-06-08 by microbit-carlos

fabien59dk, 2021-08-13:
Hello,
What should be done with the micropython-microbit-v2.0.0.hex file here : [https://github.com/microbit-foundation/micropython-microbit-v2/releases/tag/v2.0.0](url) ?
Should it be uploaded as a new firmware by following the procedure given here : [https://microbit.org/get-started/user-guide/firmware/](url)
Thanks for your help.