microbit-foundation/micropython-microbit-v2 #80: The display.get_pixel method return invalid value

ardaij, 2021-06-05:
Hi,

In our school, we programming micro:bits in MicroPython.

The display.get_pixel method return invalid values, e.g. 8, 16, 32, 64, 128, 255, instead of 0-9. What is the problem?
This discourages our kids from programming.

We use microbit v2 microcontrollers: MicroPython v1.13 on 2021-03-16; micro:bit v2.0.0-beta.5 with nRF52833

Please help us.

I also asked this question in another forum because I don’t know where we can get help.

Thanks

dpgeorge, 2021-06-05:
This was previously reported in #75, and fixed by commit 8a1dbe4fbc31d3115207aeab4af8a5f3e5aade48

I'm not sure if the newest version with the fix is available yet, @microbit-carlos ?

ardaij, 2021-06-05:
I just built the micropython from the github repository, and I copied the hex file onto the microbits.

MicroPython v1.15-64-g1e2f0d280 on 2021-06-05; micro:bit v2.0.0-beta.5 with nRF52833

The pixel values ​​are returned correctly by the get_pixel() method. This seems to have solved the problem. 

Thanks

closed 2021-06-05 by ardaij

microbit-carlos, 2021-06-07:
Glad you were able to run MicroPython from source @ardaij!
We haven't tagged and released since the fix was added to the repo, but will be present in a new release very soon.

scath999, 2021-11-26:
@microbit-carlos Wondering if you or @ardaij could run me through how you build micropython for the nrf52833? Been working on this issue for days and I'm hard-stuck.

Sorry to bug anyone, not trying to re-open this issue - only looking for a point in the right direction here.

Thanks for your patience.

microbit-carlos, 2021-12-10:
@scath999 if you've encountered any errors by running the instructions in the README then it'd be good to open a new issue to look into that.