microbit-foundation/micropython-microbit-v2 #81: Feature request: accelerometer.get_strength()

microbit-mark, 2021-06-16:
The MakeCode editor has a block for the 'strength' of the accelerometer defined in https://github.com/microsoft/pxt-microbit/blob/d80514bda223f729d452c1e51047f3272a2317fb/sim/state/accelerometer.ts#L173

[image]

It would be useful for students and those making equivalent MicroPython projects to have this API

whaleygeek, 2021-06-16:
Thanks @microbit-mark this is a small but real omission from MicroPython.

As a content writer I am finding more and more organisations are asking me to provide resources in *both* MakeCode and MicroPython, so correspondence of facilities between the two is becoming a common trend. Anything simple that either editor can do to be more aligned with the other will always be a good thing.

dpgeorge, 2021-09-06:
If added, needs to also be added in v1.

dpgeorge, 2021-09-10:
The `get_strength()` method was added in b90a60cae8fc869077b9577011455a442e70aba5

It was also added in v1.

closed 2021-09-10 by dpgeorge