microbit-foundation/micropython-microbit-v2 #82: Default "Hello world" program stalls on V2 after scrolling the text, and doesn't display the heart image.

martinwork, 2021-06-27:
Here are saved hexes: 
[microbit-program.zip](https://github.com/microbit-foundation/micropython-microbit-v2/files/6721797/microbit-program.zip)

carlosperate, 2021-06-27:
Both hex files in the zip file work fine for me. I believe both hex files were generated with the Python Editor, right?
Did you flash the hex files via BLE? Or do they file on your environment via drag&drop as well?

martinwork, 2021-06-27:
Noticed first in iOS, then created those files in Windows and dragged them to micro:bit. Tried it several times with beta and live. Very odd. I'll try a different micro:bit.

martinwork, 2021-06-27:
Sorry! It's just one micro:bit  that fails (a retail one):
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 9904360259004e450054300200000011000000009796990b
HIC ID: 9796990b
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Incompatible image detection: 1
Page erasing: 0
Daplink Mode: Interface
Interface Version: 0255
Bootloader Version: 0255
Git SHA: 1436bdcc67029fdfc0ff03b73e12045bb6a9f272
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0x828c6069
Interface CRC: 0x5b5cc0f5
Remount count: 1
URL: https://microbit.org/device/?id=9904&v=0255

This is an RC1 that works (seems identical?):
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 9904360249624e45003a601400000062000000009796990b
HIC ID: 9796990b
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Incompatible image detection: 1
Page erasing: 0
Daplink Mode: Interface
Interface Version: 0255
Bootloader Version: 0255
Git SHA: 1436bdcc67029fdfc0ff03b73e12045bb6a9f272
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0x828c6069
Interface CRC: 0x5b5cc0f5
Remount count: 0
URL: https://microbit.org/device/?id=9904&v=0255

martinwork, 2021-06-27:
I've tried some variations. If I display an image first, the image is frozen on the display. With the scrolling text first, the display is left blank.

carlosperate, 2021-06-27:
Right, and with this specific micro:bit the problem happens with BLE and drag&drop flashing?
Does the same problem happen with WebUSB flashing?

martinwork, 2021-06-28:
Yes, WebUSB is the same.

martinwork, 2021-06-29:
I guess it's a fault in the micro:bit

closed 2021-06-29 by martinwork